FreeCAD/FreeCAD #17505: Draft_Fillet gives wrong information in error message

semhustej, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the sample example, Draft_Fillet gives wrong error information when trying to fillet two objects.

Steps to reproduce:

- open sample file 
[fillet-02.zip](https://github.com/user-attachments/files/17529884/fillet-02.zip)

- select the line and the fillet objcet
- start Draft_Fillet command
- enter 2mm as a fillet radius and hit enter
- error shows up:
```
12:06:15  Edges are not connected or radius is too large.
12:06:15  Fillet cannot be created
```

This information is not valid, because when user Draft_Downgrade the fillet object to individual edges, fillet command with R=2 works without error.

See behavior in video with commentary:

https://github.com/user-attachments/assets/89250df0-9c62-40c6-a38d-3c144dfbc50c

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Draft

### Anything else?

I am able to reproduce the behavior in FC 0.21.2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-26:
@Roy-043 FYI

Roy-043, 2024-10-26:
This is a known limitation of the current tool.
https://wiki.freecad.org/Draft_Fillet#Notes

I have already prepared some code to fix this. But I was too late for the 1.0 feature freeze.

PR #17945 (2024-11-25): Draft: Allow selection of subelements for Draft_Fillet
Fixes #17505.
commit 9ec977f8 (2024-11-25): Draft: Allow selection of subelements for Draft_Fillet

Fixes #17505.
closed 2024-11-25 by yorikvanhavre