FreeCAD/FreeCAD #17507: Core: The viewer freezes when selecting vertices that are too close together

luisDias89, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I started a simple assembly, and as I added components, I noticed the software began to lag. Upon investigating, I realized that when I zoom out and the piece appears smaller, selecting it becomes problematic due to the large amount of information, causing the software to struggle with the selection and completely freeze. This is concerning because I’m only at the beginning of an assembly that will have over 1000 parts, and so far.

I'm attaching a video demonstration along with the file for analysis.
https://github.com/user-attachments/assets/b49c8694-f67c-4e35-9b9a-d2dd9c58ff52

Project:
[CNC2024_Project.zip](https://github.com/user-attachments/files/17530393/CNC2024_Project.zip)

Processaodr i7 8700 K  + NVIDIA 4060 ti

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-26:
@luisDias89 is navigation also laggy (when starting to orbit and the cursor is over geometry)?
Could be the same as reported here: https://github.com/FreeCAD/FreeCAD/issues/17185 https://github.com/FreeCAD/FreeCAD/issues/7214

luisDias89, 2024-10-26:
Navigation and orbiting are not slow. The software freezes while processing the selection of vertices. I believe this may be related to the number of vertices that can be selected within the area where the cursor is positioned. It seems the code in that section should be reviewed; it looks like it receives numerous inputs and cannot handle them, leading to a processing loop. Only when I move the cursor away does it eventually allow zooming and rotating again after some time.

Do the test yourself by downloading the file and testing it on your PC

luisDias89, 2024-10-26:
I think it's the same issue as #7214

closed 2024-10-27 by luzpaz