FreeCAD/FreeCAD #17509: SIGSEGV in BRep_Tool::Curve

antekone, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I click on my of these entries, FreeCAD crashes with SIGSEGV:

[image]

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x57980) [0x7fd2a0e57980]
#1  0x7fd27c269fc6 in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /tmp/.mount_FreeCAQBPQSR/usr/lib/./libTKBRep.so.7.7+0x26
#2  0x7fd27c27bb21 in BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) from /tmp/.mount_FreeCAQBPQSR/usr/lib/./libTKBRep.so.7.7+0xd1
#3  0x7fd27c27cd35 in BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) from /tmp/.mount_FreeCAQBPQSR/usr/lib/./libTKBRep.so.7.7+0xd5
#4  0x7fd27c8d2dce in Measure::Measurement::length() const from /tmp/.mount_FreeCAQBPQSR/usr/lib/Measure.so+0x15e
#5  0x7fd26cbd0de7 in MeasureGui::QuickMeasure::printResult() from /tmp/.mount_FreeCAQBPQSR/usr/lib/MeasureGui.so+0x947
#6  0x7fd26cbd19ee in MeasureGui::QuickMeasure::processSelection() from /tmp/.mount_FreeCAQBPQSR/usr/lib/MeasureGui.so+0x1e
#7  /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7fd2a14eb6f2]
#8  0x7fd2a14ef34a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libQt5Core.so.5+0x2a
#9  0x7fd2a14e064d in QObject::event(QEvent*) from /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libQt5Core.so.5+0x7d
#10  0x7fd2a23703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#11  0x7fd2a4169079 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libFreeCADGui.so+0x99
#12  0x7fd2a14b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libQt5Core.so.5+0x108
#13  0x7fd2a1509f51 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libQt5Core.so.5+0x381
#14  /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libQt5Core.so.5(+0x30a75c) [0x7fd2a150a75c]
#15  /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7fd29fb10ad7]
#16  /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7fd29fb13ac7]
#17  /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7fd29fb140ce]
#18  0x7fd2a150aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libQt5Core.so.5+0x66
#19  0x7fd2a14b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libQt5Core.so.5+0x11b
#20  0x7fd2a14bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libQt5Core.so.5+0x82
#21  0x7fd2a40c84e1 in Gui::Application::runApplication() from /tmp/.mount_FreeCAQBPQSR/usr/bin/../lib/libFreeCADGui.so+0xa41
#22  /tmp/.mount_FreeCAQBPQSR/usr/bin/freecad(+0x789f) [0x557a8eff089f]
#23  /lib64/libc.so.6(+0x40eec) [0x7fd2a0e40eec]
#24  /lib64/libc.so.6(__libc_start_main+0x87) [0x7fd2a0e40fb5]
#25  /tmp/.mount_FreeCAQBPQSR/usr/bin/freecad(+0x7c12) [0x557a8eff0c12]
```

[amiga_swatch_box_fcad10.zip](https://github.com/user-attachments/files/17531466/amiga_swatch_box_fcad10.zip)

### Full version info

```shell
FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311.AppImage

FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

antekone, 2024-10-26:
It *works* in Weekly version:

FreeCAD_weekly-builds-39057-conda-Linux-x86_64-py311.AppImage

(but the weekly version seems to have the "center of rotation" function broken)

closed 2024-10-26 by maxwxyz

luzpaz, 2024-10-27:
> (but the weekly version seems to have the "center of rotation" function broken)

Do you mind opening a ticket for this ? TIA