FreeCAD/FreeCAD #17511: ALIASING in the INTERFACE

MarkoAntonelli, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

System font aliasing:
in Windows 10 the system font isn't respected in the FreeCAD 1.0:
In windows the aliasing has been completely disabled; instead FreeCAD has the font.
In Attachment the magnified text inside FreeCAD 1.0 with few font aliased, and a second image with the magnified text from the file manager where the text is NOT-Aliased.

[image]
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-26:
Sounds like [qtbug 112503](https://bugreports.qt.io/plugins/servlet/mobile#issue/QTBUG-112503)

luzpaz, 2024-11-23:
Looks like there is a compiling flag that can be used as a workaround https://bugreports.qt.io/browse/QTBUG-112503?focusedId=733324&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-733324

hyarion, 2025-08-22:
@MarkoAntonelli have you tested 1.1 to see if it has been fixed there? (1.1 uses qt6 where it might have been fixed)

FEA-eng, 2026-03-08:
@MarkoAntonelli Please confirm if it's still relevant.

FEA-eng, 2026-09-12:
Should still be relevant