FreeCAD/FreeCAD #17512: Dialogue box text not readable when docked

juhulu, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working in the Part Design workbench and clicking the 'Create sketch' icon, the text for the selection of the attachment is black on a white background as long as the dialogue is docked to the side panel. When undocking the dialogue to a floating dialogue in the drawing area, the text is still black on white (desired behavior). When I dock the dialogue in the drawing area, e.g. to the right side of the drawing area, the text turns white and is not readable anymore against the white background. 

Docked in drawing area:
[image]

Docked to side panel:
[image]

Undocked in drawing area:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-27:
@MisterMakerNL FYI
@juhulu can you try is in the latest dev version https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds, also in safe mode?

MisterMakerNL, 2024-10-27:
I cannot reporduce this is my own build at windows 10 on QT6.
But this again feels like a QT style issue. Could op try the same in with the light stylesheet?

juhulu, 2024-10-27:
> @MisterMakerNL FYI @juhulu can you try is in the latest dev version https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds, also in safe mode?

Just did this - Normal mode still shows the described glitch, but it disappears when FC is started in Safe Mode.

maxwxyz, 2024-10-27:
can you upload your user.cfg file then? It seems to work with a fresh install then.

juhulu, 2024-10-27:
> can you upload your user.cfg file then? It seems to work with a fresh install then.
Renamed to user.txt:
[user.txt](https://github.com/user-attachments/files/17533720/user.txt)

MisterMakerNL, 2024-10-27:
Maybe <FCText Name="MainWindowState"> causes issues as it impacts the window states. This is used in Linkstage.

closed 2024-10-27 by maxwxyz