FreeCAD/FreeCAD #17513: Sketcher: Every attempt of adding the last constraint or build new geometry gives an error message.

3ntr0pyc0in, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have solved the problem in #17420 as you have suggested but now whatever i do in order to constraint the geometry in this sketch give an error (redundant constraint).

Applying 2mm length constraint to the diagonal line.

https://github.com/user-attachments/assets/83cd2d79-5726-4376-b45e-89e8a3079c3b

Trying to create new geometry and constrain or move it.

https://github.com/user-attachments/assets/aa0ea910-8f2a-4f54-9526-c140560cd39b

This is the file [BugN003.zip](https://github.com/user-attachments/files/17531924/BugN003.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
  * AirPlaneDesign 0.4.1
  * Assembly4 0.50.15
  * CurvedShapes 1.0.12
  * EM 2.1.1
  * ExplodedAssembly
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * OpticsWorkbench 1.0.25
  * Reinforcement
  * Rocket 3.3.0
  * sheetmetal 0.5.3
```

### Subproject(s) affected?

Sketcher

### Anything else?

This behaviour is present also in:

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-27:
The problem is caused by the first element in the sketch. It is an arc that has been turned into a full circle. That is not supported.

IMO this is a user error.

3ntr0pyc0in, 2024-10-28:
Is it possible to add a more descriptive error message (on the screen or in the error output) that explain that an arch has been closed and become a circle and that the solver cant solve it?. I'm asking because I couldn't have solved this problem if you hadn't given me your explanation.

closed 2025-09-06 by github-actions[bot]