FreeCAD/FreeCAD #17514: Extremly slow in saving dialog

brenard, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I try to save a project, or save an export, the saving dialog is extremely slow (but its works). I have the same problem with the version included in Debian Bookworm (0.20.2), the 0.20.2 Appimage version and event if with latest Appimage version (0.21.2).

[image]

After a while, I can save my project (or an export) and all start to run smoothly again.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * fasteners 0.3.46
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-27:
> Version: 0.20.2

Ancient history.
Download RC2 https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2

Closing. If issue persists for some reason, please notate ticket

closed 2024-10-27 by luzpaz

brenard, 2024-10-27:
Same problem with the 1.0.0RC2 release:
[image]