FreeCAD/FreeCAD #17515: There is a bug in ellipse creation under certain conditions.

gootaro2022, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you create an ellipse in Sketch by aligning it with the center point and try to draw it by dragging the mouse to the top, it will not be constrained correctly.
Also, entering dimensions in the same way will result in an error.

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Japanese/Japan (ja_JP)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * CurvedShapes 1.0.8
  * Curves 0.6.43
  * DesignSPHysics 0.7.0(15-09-2023)
  * EM 2.1.1
  * fasteners 0.4.70
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-27:
@PaddleStroke FYI

tback, 2024-10-29:
The file uploaded in https://forum.freecad.org/viewtopic.php?t=91733 shows a similar behavior.

knipknap, 2025-03-30:
Encountered this as well - the issue seems to be that one of the dimensions must be smaller or equal than the other one. If the solver fails, you can constrain it by swapping the dimensions and rotating the ellipse by 90°.

FEA-eng, 2025-10-25:
I can't reproduce it. @gootaro2022 or @knipknap Can you confirm if this issue is still relevant?

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```