FreeCAD/FreeCAD #17516: explodedassembly it does not work

Roberto8303, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
why do i get this error? it doesn't let me explode the flange.

### Full version info

```shell
OS: Windows 10 build 19042
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/OpenDark/Fusion
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * CamScripts
  * ExplodedAssembly
  * fasteners 0.5.29
  * FeedsAndSpeeds 0.5.0
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.9.1
  * PieMenu 1.8.0
  * ProDarkThemePreferencePack 1.0.0
  * QuickMeasure 2022.10.28
  * Render 2024.9.25
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-27:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.

- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.
- Try it in safe mode
- Add a more precise title to the issue

luzpaz, 2024-10-27:
This looks like it's an issue with the 3rd-party addon https://github.com/JMG1/ExplodedAssembly  
Closing as this ticket should be opened in that specific bugtracker

closed 2024-10-27 by luzpaz

Roberto8303, 2024-10-27:
I also tried with version 0.21 and it doesn't work as soon as I can I insert the file

luzpaz, 2024-10-27:
@Roberto8303 read up on https://wiki.freecad.org/External_workbenches (AKA addons) to understand why this ticket should be directed to the developer in charge of said 3rd-party code.