FreeCAD/FreeCAD #17517: FC1.0RC2 crashes when selecting a DatumLine

michaelrommel, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Simple model:
- one revolution
- selecting top circle as "circle" reference for a datum line
- rotating the datum line around the Y axis 45 degrees
- save file
- Clicking on the Datum Line repeatedly crashes FC

### Full version info

```shell
OS: macOS 13.7
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Render 2023.8.18
  * Defeaturing 1.2.1
  * Manipulator 1.5.0
  * kicadStepUpMod 10.21.7
  * lattice2 1.0.0
  * Curves 0.6.48
```

### Subproject(s) affected?

None

### Anything else?

[Tubeadapter.FCStd.zip](https://github.com/user-attachments/files/17533754/Tubeadapter.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-10-27:
No crash here.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39059 (Git)
Build type: Debug
Branch: master
Hash: 529779fb53085ad28bc58ece27078f0fe1a2f79d
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

xtemp09, 2024-10-27:
@michaelrommel, can you upload the crash report?

michaelrommel, 2024-10-27:
Thanks for looking into this!!

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [9863]
Path:                  /Applications/FreeCAD 2.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-10-27 13:38:27.7050 +0100
OS Version:            macOS 13.7 (22H123)
Report Version:        12
Anonymous UUID:        617D5A81-997D-E771-B6E4-B364BD10F2DC

Sleep/Wake UUID:       0E88F138-55C1-4BAE-A5AB-E93A42CA83EF

Time Awake Since Boot: 900000 seconds
Time Since Wake:       100543 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   freecad [9863]

Application Specific Information:
abort() called

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1985e4704 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x19861bc28 pthread_kill + 288
2   libsystem_c.dylib             	       0x198529ae8 abort + 180
3   libc++abi.dylib               	       0x1985d4b84 abort_message + 132
4   libc++abi.dylib               	       0x1985c43d0 demangling_terminate_handler() + 348
5   libobjc.A.dylib               	       0x19829ae68 _objc_terminate() + 160
6   libc++abi.dylib               	       0x1985d3f48 std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x1985d6d34 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 36
8   libc++abi.dylib               	       0x1985d6ce0 __cxa_throw + 140
9   Measure.so                    	       0x11868943c Measure::Measurement::length() const + 792
10  MeasureGui.so                 	       0x11e428038 MeasureGui::QuickMeasure::printResult() + 1480
11  MeasureGui.so                 	       0x11e427428 MeasureGui::QuickMeasure::processSelection() + 52
12  libQt5Core.5.15.13.dylib      	       0x106471e84 0x10632c000 + 1334916
13  libQt5Core.5.15.13.dylib      	       0x106455094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
14  CoreFoundation                	       0x1986fa500 __CFRunLoopDoSource0 + 176
15  CoreFoundation                	       0x1986fa270 __CFRunLoopDoSources0 + 244
16  CoreFoundation                	       0x1986f8e78 __CFRunLoopRun + 828
17  CoreFoundation                	       0x1986f83e8 CFRunLoopRunSpecific + 612
18  HIToolbox                     	       0x1a1f4fdf0 RunCurrentEventLoopInMode + 292
19  HIToolbox                     	       0x1a1f4fc2c ReceiveNextEventCommon + 648
20  HIToolbox                     	       0x1a1f4f984 _BlockUntilNextEventMatchingListInModeWithFilter + 76
21  AppKit                        	       0x19b92038c _DPSNextEvent + 636
22  AppKit                        	       0x19b91f528 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
23  AppKit                        	       0x19b91398c -[NSApplication run] + 464
24  libqcocoa.dylib               	       0x10add44d0 0x10ada8000 + 181456
25  freecad                       	       0x1043e3248 main + 3464
26  dyld                          	       0x1982c3fd8 start + 2412

Thread 1:
0   libsystem_pthread.dylib       	       0x198616d8c start_wqthread + 0

[... 2238 lines cut ...]
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: Drunkdeer G65 US
USB Device: BCC950 ConferenceCam
USB Device: USB2.0 Hub
USB Device: UMC1820
USB Device: DYMO LabelWriter 450
USB Device: USB2.0 Hub
USB Device: miscellaneous_device
USB Device: USB Receiver
USB Device: 2.4G Wireless Receiver
USB Device: USB Billboard Device
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Device: Thunderbolt 3 Dock, OWC, 1, 20.1
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Device: USB4.0 SSD, Intel, 1, 62.1
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Device: USB4.0 SSD, Intel, 1, 62.1
Thunderbolt Bus: Mac Studio, Apple Inc.
Thunderbolt Device: 40WP95C, LG Electronics, 1, 36.36
```

xtemp09, 2024-10-27:
A txt file or a collapsable section[^1] would be better. =)

[^1]: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/organizing-information-with-collapsed-sections

michaelrommel, 2024-10-27:
Should I rather switch to the 1.1.0dev branches? I wanted to help out with finding bugs in the Release Candidate. But If there are already fixes for some things... I do not want to waste anyone's time with outdated bg reports...

xtemp09, 2024-10-27:
Testing the latest weekly bundle will help.

michaelrommel, 2024-10-27:
OK, then I will install this and close this bug, if FC does not crash on that file on my system, too...

michaelrommel, 2024-10-27:
This is not my lucky day:

[image]

luzpaz, 2024-10-27:
@adrianinsaval :point_up_2: (is this a known macOS packaging issue?)

wwmayer, 2024-10-27:
https://github.com/FreeCAD/FreeCAD/pull/16921

adrianinsaval, 2024-10-27:
> @adrianinsaval 👆 (is this a known macOS packaging issue?)

yes, since the weeklies are not signed you need to remove apple's protection bs from the file: https://forum.freecad.org/viewtopic.php?p=778078#p778078

michaelrommel, 2024-10-27:
Oh, that again... I got different messages from this quarantine stuff before, so this didn't click with me...

```
xattr -d com.apple.quarantine /Applications/FreeCAD.app
```

removed it and now I can use the weekly build.

And I can confirm, that the click on the DatumLine no longer crashes. Thanks all for your help!!

closed 2024-10-27 by michaelrommel