FreeCAD/FreeCAD #17521: Spacemouse speed problem

APOS80, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Even though spacemouse speed has been set to its lowest in the driver settings its a bit to much.

At least when working with smaller objects.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Swedish/Sweden (sv_SE)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * BIM021 2021.12.0
  * QuickMeasure 2022.10.28
  * Trails 2022.1.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-27:
@pskowronskiTDx FYI, Is this an issue for the driver/preferences itself?

APOS80, 2024-10-27:
> @pskowronskiTDx FYI, Is this an issue for the driver/preferences itself?

I have no problem in other software but that’s on another scale, big stuff. 

With the models in freecad that’s just a decimetre in size it becomes a problem.

APOS80, 2024-10-28:
> @APOS80 You are experiencing the issue with the "Fly mode" enabled in the driver settings, right?

Yes that’s right.

luzpaz, 2025-05-10:
Is this still relevant in 1.1dev  ?

luzpaz, 2025-08-22:
@APOS80 ping. Please test on 1.1.0dev

luzpaz, 2026-01-21:
@pskowronskiTDx can you reproduce ?