FreeCAD/FreeCAD #17523: PartDesign | RC's and dev releases Color settings set back to default on reopening the material settings or editing the object

hawkeyexp, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If a color is set to an object and you reopen the view settings all is set back to default. 

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ArchTextures
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.48
  * freecad.gears 1.3.0
  * pcb 6.2023.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-27:
Should be fixed in the weekly already

hawkeyexp, 2024-10-27:
I'm sorry checked with latest weekly and still present...

OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ArchTextures
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.48
  * freecad.gears 1.3.0
  * pcb 6.2023.1

maxwxyz, 2024-10-27:
Can you provide an example file and steps to reproduce? Are you applying colors to bodies, features, parts or selected elements like faces? How is color applied?

hawkeyexp, 2024-10-27:
I created a short video cause it can be reproduced with a simple object :-)

https://drive.google.com/file/d/1Ou0Mzlh2RfQzQK5bOvS9dCe83ZvyCaek/view?usp=sharing

maxwxyz, 2024-10-28:
@davesrocketshop FYI

davesrocketshop, 2024-10-28:
Yup. I can reproduce

commit 98e7dd94 (2024-11-04): Mat: Tmp. block signals of material widget

Fixes #17523
commit 279e39bd (2024-11-04): Mat: Tmp. block signals of material widget

Fixes #17523
closed 2024-11-04 by yorikvanhavre