FreeCAD/FreeCAD #17525: Part: Projection on surface does not work in RC 1.0

DeLang1, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Projection on Surface does not work with the RC 1.0 version as with the weekly build.
Creates always this error message:

12:45:00  Part::ProjectOnSurface: Link(s) to object(s) 'Pad' go out of the allowed scope 'Projection'. Instead, the linked object(s) reside within 'Body Body'.

Freecad 0.21 does the same operation without problems.

### Full version info

```shell
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-27:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.

- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.

DeLang1, 2024-10-27:
Good evening.
Attached I add a File with the Test of a Projection.
I think the behavior of the parameters Extrude height and Solid Depth changed.
If I type in an high Extrude height (such as 20), the extrusion will be seen on the backside of the plate.
In 0.21 the Extrusion will be created on the selected plate.

Thanks for checking.

[Test Projection.zip](https://github.com/user-attachments/files/17535230/Test.Projection.zip)

sslupsky, 2024-11-13:
Here is another example of a problem attempting to project to a curved surface.

[image]

[projection problem.FCStd.zip](https://github.com/user-attachments/files/17737284/projection.problem.FCStd.zip)

sslupsky, 2024-11-13:
@DeLang1 I found with your example that if you set the Direction vector to [0, 0, -1] it appears to project and extrude on the surface of the target.

[image]

mixmid, 2025-03-27:
Part: Projection on surface still does not work in FreeCAD 1.0.0 and FreeCAD weekly-builds-40971. Tested under Linux, macOS, Windows. Just use Part: Projection on surface on the sphere and click "add face" button on "88" string, nothing happens.

The version info

```
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda AppImage
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/100.707/1
```
```

OS: macOS 14.7.4
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

[ball88.zip](https://github.com/user-attachments/files/19482454/ball88.zip)

The same but in FreeCAD 0.21, "88" is successfully projected on the sphere

[ball88_0.21.2.zip](https://github.com/user-attachments/files/19482540/ball88_0.21.2.zip)

Also this screenshot may help
```

20:19:26  Part::ProjectOnSurface: Link(s) to object(s) 'Sphere' go out of the allowed scope 'Projection001'. Instead, the linked object(s) reside within 'Body Body'.
20:20:16  Part::ProjectOnSurface: Link(s) to object(s) 'Sphere' go out of the allowed scope 'Projection'. Instead, the linked object(s) reside within 'Body Body'.
20:20:16  Part::ProjectOnSurface: Link(s) to object(s) 'Sphere' go out of the allowed scope 'Projection001'. Instead, the linked object(s) reside within 'Body Body'.
20:20:46  Part::ProjectOnSurface: Link(s) to object(s) 'Sphere' go out of the allowed scope 'Projection002'. Instead, the linked object(s) reside within 'Body Body'.
```

[image]

closed 2025-12-09 by maxwxyz