FreeCAD/FreeCAD #17526: Saved UI settings for TreeView Font size do not persist.

philjynx, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Changing TreeView font size and applying it works immediately & is also saved to (in my case) C:\Users\Phil\AppData\Roaming\FreeCAD\user.cfg. 

Quit FreeCAD. 
Start FreeCAD - font size is as it was before the changes.
Suggestion: <FCInt Name="FontSize" is not being read at startup.

<FCParamGroup Name="TreeView">
     <FCInt Name="FontSize" Value="25"/>
 </FCParamGroup>
[image]

I first notices this in RC2 and checked that it is still faulty in this version

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * Assembly4 0.50.2
  * DynamicData 2.46.0
  * ExplodedAssembly
  * fcgear
  * Help 1.0.3
  * IconThemes
  * kicadStepUpMod 10.19.4
  * Manipulator 1.5.0
  * sheetmetal 0.2.63
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

philjynx, 2024-10-27:
[image]
managed not to upload this image before.

maxwxyz, 2024-10-27:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/17061

closed 2024-10-27 by maxwxyz