FreeCAD/FreeCAD #17527: UI TreeView Icon Size only affects top level

philjynx, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Preferences-General-Display-UI-TreeView-Icon size only affects top level icons, their childrens' sizes are not altered.
The tree view _**CREATES THE VERTICAL SPACE REQUIRED BUT DOES NOT RESIZE THE ICONS**_
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * Assembly4 0.50.2
  * DynamicData 2.46.0
  * ExplodedAssembly
  * fcgear
  * Help 1.0.3
  * IconThemes
  * kicadStepUpMod 10.19.4
  * Manipulator 1.5.0
  * sheetmetal 0.2.63
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-27:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/17061

philjynx, 2024-10-29:
No, not a duplicate as it does change the Document Icon sizes but not the document's children.

Please can someone explain what 'not planned' means here? 

Why have settings for the user to alter if they have no function?

philjynx, 2024-10-29:
RealThunder link branch:

[image]
[image]

Why not merged?

Please, someone other than maxwxyz answer this.

maxwxyz, 2024-10-31:
Cannot confirm, it works as expected:
[image]

```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.12
  * Curves 0.6.50
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.25
  * Rocket 3.3.0
  * sheetmetal 0.5.3
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

luzpaz, 2024-10-31:
@philjynx can you change to dark or light theme in the preferences to see if this changes anything ?

philjynx, 2024-10-31:
@luzpaz Hi, I have tried lots of variations. None of them produces uniform sized treeview icons.
I'm getting the same results with both Version: 1.1.0dev.39057 (Git) and Version: 1.0.0RC2.38806 (Git)

I _**am getting**_ useable font sizes [https://github.com/FreeCAD/FreeCAD/issues/17526](url) with this configuration:

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) **AND** Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fcgear
  * FreeCAD-themes 2024.7.24
  * Help 1.0.3
  * HiResMonitor
  * kicadStepUpMod 11.3.9
  * Manipulator 1.5.7
  * sheetmetal 0.5.3

HiResMonitor contains InitGui.py found at the shown URL

`#https://forum.freecadweb.org/viewtopic.php?f=3&t=9662&start=10
#creer un repertoire "HighRezMonitor" dans "C:\Users\xxxx\AppData\Roaming\FreeCAD\Mod"
#creer un fichier "InitGui.py"
#collez le code:
#
import PySide
from PySide import QtGui
f=QtGui.QApplication.font()
f.pointSize()                   # returns e.g. 8
f.setPointSize(16)              #nouvelle dimension
f=QtGui.QApplication.setFont(f)`
[image]

philjynx, 2024-10-31:
Example of full propogation of font size and incomplete propogation:
[image]

[image]

hasecilu, 2024-11-04:
Also changed to 48 px but icons were not enlarged

[image]

```
OS: Arch Linux (XFCE/openbox/xcb)
Architecture: x86_64
Version: 1.1.0dev.36460 +2653 (Git)
Build type: Release
Branch: test/prs
Hash: 440b96eaf3a2a12a9079b3944101a11cb53a25fe
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
```

skewbdew, 2024-12-20:
Toggling 'show visibility icon' and changing the icon size seems to fix it for the rest of the session

hasecilu, 2024-12-21:
When you make the icon smaller the update is done properly but making it greater fails

philjynx, 2024-12-21:
Sort of fixes it for session but it's inconsistent even then.
[image]

FEA-eng, 2025-10-25:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```