FreeCAD/FreeCAD #17529: PartDesign DAE export fails when SubShapeBinder of sub-component is made

Jarno-de-Wit, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Exporting a PartDesign body fails if a (sub)shapebinder is made of any component of the body.

This is caused by the use of [Arch.PruneIsIncluded](https://github.com/FreeCAD/FreeCAD/blob/c5c64a9c70f21c4c4c2ad294645993f9a60342d8/src/Mod/BIM/ArchCommands.py#L749-L795) in [importDAE.py](https://github.com/FreeCAD/FreeCAD/blob/c5c64a9c70f21c4c4c2ad294645993f9a60342d8/src/Mod/BIM/importers/importDAE.py#L208). The intention of calling this function is to filter out objects which are components of other objects, likely to filter out e.g. if the user accidentally selected a component of a body instead of the body.

To accomplish this, the PruneIsIncluded function checks the objects' InList to see if it is a part of another object. While this approach might work fine for some scenarios, this approach can be too rigorous for PartDesign Bodies. If a subShapeBinder is made of any component of a PartDesign Body, this SubShapeBinder is also included in the Body's InList. As such, when filtering the selection during the export, this Body will also get filtered out, even if the only dependency is a SubShapeBinder of a sketch inside the body.

## Proposed fixes:
Firstly, the pruning process should be adjusted to be more careful about when to drop an object from a selection. In Arch.PruneIsIncluded the `PartDesign::SubShapeBinder` should be included in the [list of allowed types](https://github.com/FreeCAD/FreeCAD/blob/c5c64a9c70f21c4c4c2ad294645993f9a60342d8/src/Mod/BIM/ArchCommands.py#L762), since this type is usually a fairly clear indication that the user still cares about the original object.
Secondly, dropping an object from a selection should not be done silently. As such, when an object is deemed not to be "toplevel" in Arch.PruneIsIncluded, it should issue a warning that an object from the selection is dropped.

### Full version info

```shell
OS: macOS Mojave (10.14)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Render 2024.9.25 (Disabled)
  * A2plus 0.4.68
  * Curves 0.6.50
```

### Subproject(s) affected?

PartDesign

### Anything else?

## Example file:
The issue can be seen in the following file:
[ExportDAETest.FCStd.zip](https://github.com/user-attachments/files/17535199/ExportDAETest.FCStd.zip)In this file, exporting the Slot object to DAE works as intended, generating the correct 3D model. Exporting the Pin object on the other hand, generates an empty DAE file, with nothing more than the standard headers.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-10-29:
Totally agree with your fixes.

Indeed that exporter (and the PruneIncluded function) are old were not really thought for PartDesign... Definitely an update there is welcome :)

Are you proposing to work on those fixes yourself? That would be great!

Jarno-de-Wit, 2024-10-29:
I can certainly look at it, and probably have a fix for it before the end of the week. The fix itself shouldn't be all that difficult, but I would like to investigate if there are any other operations (beside making SubShapeBinders) that trigger the same issue, and can thus be patched alongside this one.

I also found that a similar issue occurred in the WebGL exporter rewrite, which resulted in the use of Arch.pruneIncluded simply being commented out altogether in [9b1a8d6](https://github.com/FreeCAD/FreeCAD/pull/4025/commits/9b1a8d6d6989e46263c3213f6e9fddfea3407a6a), but I'd figure that's more of a quick patch to make that exporter work rather than an actual solution.

PR #17654 (2024-11-05): Fix excessive selection pruning in dae exporter for PartDesign Objects (#17529)
Fixes the excessive pruning of the export selection inside Arch.PruneIncluded. This was caused by the function not checking correctly for (Sub-)ShapeBinders when determining whether an object was a top level object or not, affecting the DAE exporter, and legacy IFC and OBJ exporters.

The fix includes adding a check for these object types, as well as enabling "strict" mode for PruneIncluded, which makes the pruning only take place if the parent object is also included in the selection. This latter aspect of the fix [... cut]
closed 2024-11-05 by yorikvanhavre