FreeCAD/FreeCAD #17531: PD after adding a PD_Boolean Manipulator WB stops working

ElPatoNoEsTu, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Given a FC File with two Bodys each containing a Sketch defining just some lines as refences to allign to and a sketch defining a shape which is padded. 

Unsing the Manipulator Workbench the both bodies can be alligned using the Manipulator WG flawlessly UNTIL both bodies get cloned and fused using PD_Boolean 

It seems that the Manipulator-WB gets confused by the new created boolean.  

Good:  Manipulator-WB can be used to allign Bodies 
[image]
[Alignment_Test_2.zip](https://github.com/user-attachments/files/17535766/Alignment_Test_2.zip)

Bad:  ...  The bodies were clonded an fused - Unsing the Manipulator-WB does not align the original bodies as befor even if one is aktive and the fused Body is somehow manipultated and jumps around 
[Alignment_Test_3.zip](https://github.com/user-attachments/files/17535767/Alignment_Test_3.zip)
 
Maybe the Manipulatior WB gets confused due to the data-stuctures which are created by the PD_Boolean ? 

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Curves 0.6.50
  * DynamicData 2.68.0
  * freecad.gears 1.3.0
  * LCInterlocking
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-28:
Please post this ticket to https://github.com/easyw/Manipulator 3rd-party addon repo. TIA

cc @easyw

ElPatoNoEsTu, 2024-10-28:
Hi All,   

Sorry to bother again ... I assume thats something somewhat "deeper"  or due to the reuse of alignment code (?):

befor / while align:
[image]

after align & enable visibility of the clones.

[image]

As an Idea -  how does the code handle the alignment if one Body is sketched on XY-Plane & padded  - while the other Body is sketched on the YZ-Plane & padded  ( ... does alignment quitly assumes the to align objects beeing orinented in matching LCS ? )  

Regards,  EpNeT

ElPatoNoEsTu, 2024-10-28:
@luzpaz  is there any way to reopen the issue ?  b'cause its not only the external WB its also the internal aligment tool ( see above ) ...  ?!

luzpaz, 2024-10-30:
Thanks for the ping and deeper sleuthing. Would you mind updating the ticket title to more accurately represent the issue? TIA

ElPatoNoEsTu, 2024-10-30:
... Where / how does the align-tool obtain/evaluate the TN(?) and placement of the objects to manipulate - does the code take care of/can handle the tnp mods ? ( as I notice in Github )  the code for the Manipulaor WB  which I expect beeing the predecessor of th align tool seems to be rather old ( 7 jears (?) )  ?

FEA-eng, 2025-10-28:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```