FreeCAD/FreeCAD #17533: Part: BooleanFragments: Regression: Part switches from Standard to CompSolid on document re-load + recompute

AIRCAP, 2024-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a document originally created with 0.21.1  It has a BooleanFragments part in it.
When I open this document in 1.1dev - even though it has previously been saved in 1.1dev it claims that the document needs to be updated.

If I do so and then trigger a complete recompute, the BooleanFragments feature switches from "Standard" to "CompSolid" - which breaks subsequent features for FEM analysis since the CompSolid has the wrong structure.

Expected Outcome:
BooleanFragments set to Standard stays in Standard

Actual Outcome:
BooleanFragments somehow switches to CompSolid when the document is closed and re-opened.

I am currently  trying to reproduce this issue with a smaller document for reproduction that I can share...

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39036 +7 (Git)
Build type: Debug
Branch: regression
Hash: 204a86f097dd58063c178b67729574b2b0e564b8
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.12
  * CfdOF 1.27.10
  * Curves 0.6.48
  * AirPlaneDesign 0.4.1
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-10-28:
Addition: I managed to trigger the switch with just changing a sketch, which triggered an automatic recompute of the whole document - and it switched again from Standard to CompSolid. So document re-open was not needed just a recompute

AIRCAP, 2024-10-28:
backport should be fine,  I don't think this is related, it must have to do something with file saving and loading and automatic conversion. i can only reproduce the issue with one masisive file which has lots of 3rd party extensions in it. and not with anything small. still looking into it.
I'm 99.9% certain it has absolutely nothing to do with my PRs

AIRCAP, 2024-10-28:
One notable thing about this document is, every time I open it I get:

```
Some document(s) require recomputation for migration purposes. It is highly recommended to perform a recomputation before any modification to avoid compatibility problems.

Do you want to recompute now?
```

I then do the recomputation, it usually succeeds, I save the file - and next time I open it I get the same message again. There might be something wrong with that file and that is what triggers this, but I don't know what

adrianinsaval, 2024-10-28:
> I'm 99.9% certain it has absolutely nothing to do with my PRs

does it happen in rc2?

AIRCAP, 2024-10-28:
rc2 can't recompute my file atm because of bugfixes added later. i am trying to isolate the feature a bit more then i can test that

AIRCAP, 2024-10-28:
[CFDOfbroken.zip](https://github.com/user-attachments/files/17546958/CFDOfbroken.zip)

here is the reproduction file.  if you open that file - there should be a broken empty CfdOF analysis container in it (which wasn't initially broken, but it broke when i cleared the file of unnecessary stuff)

if you have the CfdOF extension installed and open the file, the BooleanFragments container changes mode on open (or triggered recompute) to "CompSolid" 

if the broken CfdAnalysis Container is removed or CfdOF deinstalled, this doesn't happen and the BooleanFragements stays "Standard"

This is either a very weird bug where the failure to initialize the CfdOF Analysis container somehow triggers weird behaviour when the BooleanFragments is initialized

Or this is some weird crap CfdOF does. I only had that since a few days ago, so this must have triggered by something when upgrading the file, but with this file it happens even in 0.21.2

AIRCAP, 2024-10-28:
Crosslink: https://github.com/jaheyns/CfdOF/issues/190

AIRCAP, 2024-10-28:
It's not a bug, its a feature *cough*
https://github.com/jaheyns/CfdOF/blob/280bf7001dcdfba6eb0508fde9cb8d1fdfae5b4e/CfdOF/CfdTools.py#L306-L315

so not a FreeCAD issue - 3rd party CfdOF issue.

AIRCAP, 2024-10-28:
and they fixed it, so closing

closed 2024-10-28 by AIRCAP