FreeCAD/FreeCAD #17539: Concave edge selection do not work

mgth, 2024-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Concave edges inside the part seams impossible to be selected while I have no problem without convex ones.
here on the picture the edge between the blue and green face is impossible to click on.
going to wire view solve the problem.

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly3 0.12.2
  * CurvesWB 0.6.12
  * ExtremeProDark 2.6.4
  * OpenDark 2024.3.13
  * parts_library
  * parts_library.bak
  * workfeature
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-28:
Can you share the file and steps how to reproduce? What view style are you using?

mgth, 2024-10-28:
Sorry, I made changes to the file and the problem is not present anymore unfortunately I did not save it at this stage. But it was not the first time I got the problem. I will post a file if the problem is there again. It appended in view style 4 to 7 but not in wire style. maybe because there is no face to conflict with edge selection.

closed 2024-10-28 by maxwxyz

mgth, 2024-10-28:
Here is a file where the problem occurs again :
[Freecad_Bug_#17539.zip](https://github.com/user-attachments/files/17544319/Freecad_Bug_.17539.zip)

maxwxyz, 2024-10-28:
Can confirm. Workaround is to use wireframe view mode.
Could you add your findings to https://github.com/FreeCAD/FreeCAD/issues/10308 as it seems to be similar to your issue. 
Would also be solved if https://github.com/FreeCAD/FreeCAD/issues/14530 is implemented.