FreeCAD/FreeCAD #17542: UI/UX Cropped Title on popup while checking geometry with threads.

leoheck, 2024-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Part Design > Check geometry (threaded) popup, needs extra horizontal space.

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git) AppImage
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yomgui1, 2025-03-17:
I think it's not a bug, even not something we can change in FreeCAD or QT, it's a normal behaviour of the OS window manager.
The title is in the frame of the window, it's not a Qt widget but created and drived by the OS window manager.
The truncation is due to many things:
- large title font
- window widgets

Here a test with an hacked FreeCAD binary to set a very long title and another window manager (it's I3, without widgets on window frame):

[image]

You can see I've also a truncated title ending with "...".

xtemp09, 2025-05-12:
I think this problem won't be fixed.

leoheck, 2025-05-12:
Why the window is so small horizontally? It could have a minimum size, or it could be increased a bit.

xtemp09, 2025-05-12:
> it could be increased a bit.

The problem seems to come from _OCCT_.

MisterMakerNL, 2025-05-21:
Not an issue on windows:

[image]

leoheck, 2025-05-21:
Many pop-up windows are cropped and smaller than they should be on Linux.

luzpaz, 2025-08-29:
@johnocfii is this true for macOS as well ? If you don't mind checking ?

JohnOCFII, 2025-08-29:
> [@JohnOCFII](https://github.com/JohnOCFII) is this true for macOS as well ? If you don't mind checking ?

It fits fine on macOS.  My results happen so quickly, I can't get a screen shot.

```
OS: macOS 15.6.1
Architecture: arm64
Version: 1.1.0dev.43129 (Git)
Build date: 2025/08/28 22:14:55
Build type: Unknown
Branch: main
Hash: 78a579804cdc784cd6f84d70b487aa256c27437e
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.8.23
  * manifest.json
  * Telemetry 1.0.2
```

FEA-eng, 2026-09-15:
@leoheck Please confirm if it's still relevant.