FreeCAD/FreeCAD #17543: PartDesign Features now have issues with 1.1.0dev.39057, and there is no clue why this happened and how to solve them.

leoheck, 2024-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The latest change of my design using the latest Appimage (`1.1.0dev.39057`) started to give me issues on a Part Design Body.

Checking the geometry of this Body I have these results.

[image]

[image]

This is the first feature of the Body that has issues. It is the 3rd feature a Fillet, after a Pad and a Thickness. 

[image]

[image]

The issue is that everything looks good, visually and the design quite simple.

The first problem is that I have no clue how this issue happened, is it something broken on this version of Freecad?

The second problem is that I have no clue on how to tackle this thing. The Check Geometry could give us hints on what we could do to fix the issues specially when working with the Part Design WB. Each problem could have a link to a wiki describing how to fix that particular thing. 

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git) AppImage
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-28:
@leoheck please share an example file.
@AIRCAP could you take a look at this? I guess it's related to your PRs.

leoheck, 2024-10-28:
Sure @maxwxyz I will try to move the problematic object to another file

leoheck, 2024-10-28:
When I remove all the features after the broken one, the whole thing does not give me issues anymore.

leoheck, 2024-10-28:
I cannot share the file since I could not recreate a reduced version of it.

But I can share a picture, and it looks like it is possible to be related with what you said about the tolerances.
[image]

I could track a point in the file where this started to happen. It is funny because if I keep the original nested tree.. the issue appears starting from the 3rd feature (of lets say 40 or so)... but when I start deleting features from the bottom up... including Sketches.. this issue appears for the first time after the feature 10. 

I found a place where a rib (rectangle) is added to a boss (cylinder) looking from the top.. and this rectangle has 2 points on the cylinder... but the cylinder and the rectangle are small... so there is no much mass overlapping. 

I am trying to improve this Sketch to check if the whole Body can be fixed. But for now I don't have much to share yet.

The only thing I can say is that the process to track and fix this is not trivial. It looks like there is no easy/simple or helpful tool for that.

AIRCAP, 2024-10-28:
> I could track a point in the file where this started to happen. It is funny because if I keep the original nested tree.. the issue appears starting from the 3rd feature (of lets say 40 or so)... but [...]

I noticed similar. Often when I try to simplify a file for reproduction of an issue, I only leave the direct ancestors of the problematic operation - but even the simple act of "Part->Convert to Solid" which should do nothing but save the Shape of a feature as a non-parametric standalone feature introduces ever so tiny changes due to the way the coordinates are now explicitly stored as a single float as opposed to computed in a series of operations. This **should** not change the values, but in practice it does and some of the more finnicky geometry operations in OCCT are susceptible to such minute differences that an operation that previously worked now fails - or the other way around.  Extremely frustrating.

The only save thing is to avoid operations that are that "fragile" - but then one cannot use "thickness" and "3d offset" at all, which is very limiting.

AIRCAP, 2024-10-28:
I can think of a number of ways to change the rib to probably work. One is to have it intersect all the way to the center of the cylindrical part, and later maybe cut out the parts that are unneeded if that makes it too large.

But the issue is really which change we did makes geometry that previously worked now fail. For this we need an actual example/test file that previously worked and now doesn't so we can exectue the code line by line and see where there's a difference in what is computed (overly simplified)
Without a file that actually reproduces the issue (and its absence on an older FreeCAD version) there's little we can do but file this as an OCCT limitation :(

AIRCAP, 2024-10-28:
@leoheck can you try the following:
1. check if the error also appears in 1.0 RC2 ? You can use the AppImage downloads to test older version. Is this a regression in the latest weekly or was it already broken before?
2. if its only broken in the latest weekly, can you go to
**Tools->"Edit Parameters"->BaseApp->Preferences->Mod->Part->Boolean**
And then right click in the empty space underneath "Refine Model" and add a "float item" called "BooleanFuzzy" and set that to 0.0

should look like that:
[image]

then close freecad and reopen it and try again. Does that change anything?

leoheck, 2024-10-28:
> I can think of a number of ways to change the rib to probably work. One is to have it intersect all the way to the center of the cylindrical part, and later maybe cut out the parts that are unneeded [...]

Yeah, I can also think other ways of doing that. The problem is that this project/file is getting huge and slow. 
So every try takes time... and it looks like there are problems on other features that is really hard to track. 
I am about to create a new body and star moving/reorganizing the whole part. This may improve the overall structure of this model.

leoheck, 2024-10-28:
> @leoheck can you try the following:
> 
> 1. check if the error also appears in 1.0 RC2 ? You can use the AppImage downloads to test older version. Is this a regression in the latest weekly or was it already broken before?
> 2. if its only broken in the latest weekly, can you go to
>    **Tools->"Edit Parameters"->BaseApp->Preferences->Mod->Part->Boolean**
>    And then right click in the empty space underneath "Refine Model" and add a "float item" called "BooleanFuzzy" and set that to 0.0 , then "**Save to Disk**"
> 
> should look like that: ![image](https://private-user-images.githubusercontent.com/32105268/380863949-ff85ed9c-c4b5-446d-9a3d-24d0abc5ebe8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAxNDQzODcsIm5iZiI6MTczMDE0NDA4NywicGF0aCI6Ii8zMjEwNTI2OC8zODA4NjM5NDktZmY4NWVkOWMtYzRiNS00NDZkLTlhM2QtMjRkMGFiYzVlYmU4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI4VDE5MzQ0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlkZjlhNWQ3NmVkYzIxNWVhZTMwMmE5OTg1NGYxN2Q2ODk0ZDc5ZmQ4YTEyMDI1ZjQ4Mzk1ZWVmMjlkNjJlZmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.p_zn22DSDQtTDHSiJbU1r905xu4RfsDjipC5zIPVC3M)
> 
> then close freecad and reopen it and try again. Does that change anything?

Sure, let me get this AppImage. I am not sure I still have a working file at this point.

AIRCAP, 2024-10-28:
important note, when we are done with this, remember to remove that BooleanFuzzy entry again. Otherwise it will remain in your config, and - if that feature is included in 1.0 - files that work for other people with Fuzzy set to default 10 might fail to compute on your installation.

AIRCAP, 2024-10-29:
@leoheck  any news? Did you get it to work with the older version?
link to the different app image versons for different operating systems is here https://github.com/FreeCAD/FreeCAD-Bundle/releases

leoheck, 2024-10-30:
@AIRCAP a quick test showed me the same issue with 1.0-RC2. But it was not failing like that before version 1.1*. But I don't have extra info since it is hard to track accurately from where the issues come. I saw another issue posted 4 days ago that could be related too. This Body in particular have some features that were moved to another level of the Body... so features and sketches may not be in the right order. Also, some Sketches were created using a tip set in a different feature that was not the last one at that moment. There are many weak points that may help to create or increase this problem.

chennes, 2024-10-30:
> @AIRCAP a quick test showed me the same issue with 1.0-RC2.

The plot thickens -- so this cannot be due to the changes in https://github.com/FreeCAD/FreeCAD/pull/17119 since it is not *in* RC2 (and has in fact not been backported at all). So it seems that we may be in the clear when it comes to merging https://github.com/FreeCAD/FreeCAD/pull/17325 (the 1.0 port of #17119). Do I understand correctly, @AIRCAP ?

AIRCAP, 2024-10-30:
@chennes yes, thats my understanding too. @maxwxyz ping, as promised.

@leoheck thanks for the feedback. If you can share any file that exhibits this issue we can still look into it, but without it its gonna be tricky. I suspect the toponaming refactoring changed it ( simply cause that also triggered a numberbof other regressions) but we'd need a test file for a proper bisect of the git tree, unless you want to do that yourself.

leoheck, 2024-10-30:
@AIRCAP I have issues like this on every feature of the problematic model. So maybe related with the TPN stuff.

[image]

I know the feature change that triggered this issue. It was a reduction of a pad (a screw boss). Nothing problematic at all. But it triggered the recalculation of the features.

I have my files syncing with Dropbox. So I think I can go back in time where this model was not breaking to double check with 1.0-RC2 but I did not do that yet. I will try to find time to do that.

luzpaz, 2024-10-30:
> One absolutely crucial thing when testing with the same file in different versions is to always do a full recompute (select everything, mark as recompute and recompute). The automatic recompute is often insufficient and there are differences with TNP fixes plus the recent boolean/tolerance stuff.

We should document this somewhere, @maxwxyz

maxwxyz, 2024-10-30:
Agreed. But sometimes the recompute also breaks older files, so not sure how to phrase this.

adrianinsaval, 2024-10-31:
Manually making everything recompute is also confusing to do, would be nice if we had a "recompute everything" command, this off topic here though. Could someone create an issue about making recompute easier and having guidelines for testing models? Ouw comments here should probably be marked as off topic

maxwxyz, 2024-10-31:
You can right click the document and mark all for recompute

AIRCAP, 2024-10-31:
> Manually making everything recompute is also confusing to do, would be nice if we had a "recompute everything" command, this off topic here though. Could someone create an issue about making [...]

ust saying - clicking on "new issue" and making a feature request like that would have taken marginally more effort than writing this here. let me quickly do that for you: I right click on the "..." in your comment and click "reference in new issue" add literally one sentecne with a suggestion ... and voila #17589

adrianinsaval, 2025-05-21:
without any information on how to reproduce this I don't think there's anything we can do about this one, might be better to just close it