FreeCAD/FreeCAD #17548: Part Design - Additive_Loft command bug

vasik12, 2024-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello, 
on this link: https://forum.freecad.org/viewtopic.php?p=782319#p782319 is solved a problem with Additive_Loft. But RC2 v. 1.0.0 still includes this bug. I don't know if developers know about this...

### Full version info

```shell
OS: Windows 11 build 22000
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-28:
@vasik12 please test on the dev version and report back if it is still an issue. https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

closed 2024-10-28 by maxwxyz

vasik12, 2024-10-28:
> @vasik12 please test on the dev version and report back if it is still an issue. https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Hello, in this dev. is this tool all right. 
BTW. I trying import a project made in v.0.21.2 and has several other errors