FreeCAD/FreeCAD #17552: Tree View (Model) unreadable when docking the Model tab after it was undocked

Buffalo1John, 2024-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If the Tree View is undocked and then docked again, the result is that parts of the tree are unreadable. There is no change after exiting and restarting/reloading.

Here is the undocked Model window:
[image]

Here is the redocked Model window:
[image]

I had placed the cursor over the Body and you can barely see the faint light blue background when I did the PrtSc capture. The name of the file is not shown at all. I tried changing all the things regarding the top of the panel and could not make it normal.

By chance I took a look at the panels (View then Panels) and I see this (had to use my phone to take an image so I could capture the menu drop down):
[image]
So - now I have two of the Model panel?

Now if I let the menu go away, the Model panel disappears and so I checked both of the Model panels on and I get this (I also selected an item in the tree so you can see the other funky display text):
[image]

I went into the preferences and tried reseting all the general and display settings and no difference. This happens on any and every design file.

I have unzipped RC2 into it's own directory - didn't install, maybe that is related?

It's really messed up and I can't get it to a usable state...

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * lattice2 1.0.0
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-29:
@MisterMakerNL FYI

MisterMakerNL, 2024-10-29:
Wrong overlay_stylesheet is applied here, it is the dark version you can see it by the dark headers. Not sure how this got applied.

Buffalo1John, 2024-10-29:
> Wrong overlay_stylesheet is applied here, it is the dark version you can see it by the dark headers. Not sure how this got applied.

I have no clue how it changed as I have never changed styles knowingly. I thought it was related to the two instances of the Model Panel that appeared, which happens after I undock and then redock the Model Panel...

maxwxyz, 2024-10-29:
Is this the same as https://github.com/FreeCAD/FreeCAD/issues/16425 ?

MisterMakerNL, 2024-10-29:
> Is this the same as #16425 ?

Yeah. All this should be set by https://github.com/FreeCAD/FreeCAD/blob/4f323f9580334de25192d2bf1f272206ad2d3f55/src/Gui/PreferencePacks/FreeCAD%20Classic/FreeCAD%20Classic.cfg#L292

maxwxyz, 2024-10-29:
Duplicate of #16425

Buffalo1John, 2024-10-29:
What about the duplicate Model Panels? 

Should I file the duplicate Model Panels as a new bug?

maxwxyz, 2024-10-30:
I cannot reproduce this, can you try restarting in save mode (Help>Restart in safe mode) and check if the issue persists?

Buffalo1John, 2024-10-30:
RC2 doesn't have safe mode so I downloaded 39057. 

Yes, 39057 does show the extra entry in the View Panels and I got it to make a extra Tasks. I will do it again and see if I can take images to show the steps to reproduce. I will post when I have more...

Buffalo1John, 2024-10-30:
OK, first off, I started in safe mode first thing I notice is that none of the three panels (Model, Tasks, Report) are docked

Here is before restarting in safe mode:
[image]

Here is restarting in safe mode:
[image]

Here is View Panels:
[image]

Now, lets dock the Model panel:
[image]

That worked ok, checked View Panels and nothing amiss. Now undock it and then dock it again:
[image]

Now we have two Model panels. So, without fussing any more with Model, lets undock Report:
[image]

Now we have two Report after only undock
[image]

Now we will from here move the Tasks panel to dock it, checked View Panels, still had two Report, so undock Tasks and now we have two Tasks
[image]

Since we have two Tasks, lets make both visible and see what happens:
[image]

The second Tasks panel has a different look. It didn't come up on the same monitor as the FreeCAD monitor, so I moved it over on top so it is visible.
[image]

Well, lets try doing a task, I press New and I see this:
[image]

and those messages in the Report window are:
09:30:20  Migrating Start Workbench to Start command... done.
09:52:45  QMainWindowLayout::tabPosition called with out-of-bounds value '0'
09:52:51  QMainWindowLayout::tabPosition called with out-of-bounds value '0'
09:52:54  QMainWindowLayout::tabPosition called with out-of-bounds value '0'
09:53:07  QMainWindowLayout::tabPosition called with out-of-bounds value '0'
09:53:45  QMainWindowLayout::tabPosition called with out-of-bounds value '0'

So it has been complaining for a while, but the Report panel was hidden.

Buffalo1John, 2024-10-30:
Noticed something I reported wrong, the Task panel was docked when I entered Safe Mode, sorry... Everything else was as reported.