FreeCAD/FreeCAD #17553: Gui: Transaction interaction

pieterhijma, 2024-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Based on what I learned from @wwmayer and #17475, I noticed that it is quite easy to cause interactions between transactions that result in problems. One example is the following:

1. Create a new document
2. Switch to Part and create a cube
3. Move Part Design and create a sketch

At this point, a task is opened that has a pending transaction. This can be verified with:

``` python
App.ActiveDocument.HasPendingTransaction
```

4. Rename the label of the cube

This rename adds a transaction as well and now if you execute the above, the active document has no pending transaction anymore.

5. Choose a plane in the 3D view and close the task
6. Undo the rename
7. Undo creating a new sketch

Notice that the sketch is not deleted, probably because of the interfering transaction.

8. Undo the cube creation
9. Delete the Sketch
10. Undo deleting the sketch

We now have a segfault:

<details>
<summary>stacktrace:</summary>
<pre>
#0  0x0000007800700030 in ?? ()
#1  0x00007ffff3e82589 in App::PropertyLinkSubList::verifyObject (this=0x55555df569e0, obj=0x55555ed10130, parent=0x55555df55ff0) at /home/pieter/current/fpa/freecad-main/src/App/PropertyLinks.cpp:1970
#2  0x00007ffff3e82f28 in App::PropertyLinkSubList::setValues (this=0x55555df569e0, lValue=..., lSubNames=..., ShadowSubList=...) at /home/pieter/current/fpa/freecad-main/src/App/PropertyLinks.cpp:2085
#3  0x00007ffff3e82e1e in App::PropertyLinkSubList::setValues (this=0x55555df569e0, lValue=std::vector of length 1, capacity 1 = {...}, lSubNames=std::vector of length 1, capacity 1 = {...}, ShadowSubList=...) at /home/pieter/current/fpa/freecad-main/src/App/PropertyLinks.cpp:2076
#4  0x00007ffff3e88715 in App::PropertyLinkSubList::Paste (this=0x55555df569e0, from=...) at /home/pieter/current/fpa/freecad-main/src/App/PropertyLinks.cpp:2788
#5  0x00007ffff3dcb8ec in App::TransactionObject::applyChn (this=0x55555aac6ee0, pcObj=0x55555df55ff0) at /home/pieter/current/fpa/freecad-main/src/App/Transactions.cpp:351
#6  0x00007ffff3dcaace in App::Transaction::apply (this=0x55555dd80990, Doc=..., forward=false) at /home/pieter/current/fpa/freecad-main/src/App/Transactions.cpp:176
#7  0x00007ffff3b38ba8 in App::Document::undo (this=0x55555bd2c7f0, id=0) at /home/pieter/current/fpa/freecad-main/src/App/Document.cpp:231
#8  0x00007ffff58dfa13 in Gui::Document::undo (this=0x55555d2224f0, iSteps=1) at /home/pieter/current/fpa/freecad-main/src/Gui/Document.cpp:2590
#9  0x00007ffff5fa2b71 in Gui::View3DInventor::onMsg (this=0x7fffdc008590, pMsg=0x7ffff6274434 "Undo", ppReturn=0x0) at /home/pieter/current/fpa/freecad-main/src/Gui/View3DInventor.cpp:392
#10 0x00007ffff57ed64f in Gui::Application::sendMsgToActiveView (this=0x7fffffffccc0, pMsg=0x7ffff6274434 "Undo", ppReturn=0x0) at /home/pieter/current/fpa/freecad-main/src/Gui/Application.cpp:1141
#11 0x00007ffff5a30e56 in StdCmdUndo::activated (this=0x555555d4ab10, iMsg=0) at /home/pieter/current/fpa/freecad-main/src/Gui/CommandDoc.cpp:1023
#12 0x00007ffff5a11257 in Gui::Command::_invoke (this=0x555555d4ab10, id=0, disablelog=true) at /home/pieter/current/fpa/freecad-main/src/Gui/Command.cpp:444
#13 0x00007ffff5a10f71 in Gui::Command::invoke (this=0x555555d4ab10, i=0, trigger=Gui::Command::TriggerAction) at /home/pieter/current/fpa/freecad-main/src/Gui/Command.cpp:408
#14 0x00007ffff59edcfa in Gui::Action::onActivated (this=0x555558ca72f0) at /home/pieter/current/fpa/freecad-main/src/Gui/Action.cpp:111
#15 0x00007ffff5a03ab1 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(void (Gui::Action::*)(), Gui::Action*, void**)::{lambda()#1}::operator()() const (__closure=0x7fffffffb260) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:152
#16 0x00007ffff5a04eeb in QtPrivate::FunctorCallBase::call_internal<void, QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(void (Gui::Action::*)(), Gui::Action*, void**)::{lambda()#1}>(void**, QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(void (Gui::Action::*)(), Gui::Action*, void**)::{lambda()#1}&&) (args=0x7fffffffb410, fn=...) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:65
#17 0x00007ffff5a03b23 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(void (Gui::Action::*)(), Gui::Action*, void**) (f=&virtual Gui::Action::onActivated(), o=0x555558ca72f0, arg=0x7fffffffb410) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:151
#18 0x00007ffff5a02412 in QtPrivate::FunctionPointer<void (Gui::Action::*)()>::call<QtPrivate::List<>, void>(void (Gui::Action::*)(), Gui::Action*, void**) (f=&virtual Gui::Action::onActivated(), o=0x555558ca72f0, arg=0x7fffffffb410) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:199
#19 0x00007ffff5a004f7 in QtPrivate::QCallableObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=1, this_=0x55555a5f3db0, r=0x555558ca72f0, a=0x7fffffffb410, ret=0x0) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:570
#20 0x00007fffefbb2229 in QtPrivate::QSlotObjectBase::call (this=<optimized out>, r=<optimized out>, a=<optimized out>, this=<optimized out>, r=<optimized out>, a=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobjectdefs_impl.h:486
#21 doActivate<false> (sender=<optimized out>, signal_index=<optimized out>, argv=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobject.cpp:4120
#22 0x00007ffff0566524 in QAction::triggered (this=<optimized out>, _t1=<optimized out>) at /usr/src/debug/qt6-base/build/src/gui/Gui_autogen/include/moc_qaction.cpp:480
#23 QAction::activate (this=0x55555a5bca80, event=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/gui/kernel/qaction.cpp:1104
#24 0x00007ffff0c1ea75 in QAbstractButtonPrivate::click (this=0x555558ca6e50) at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qabstractbutton.cpp:364
#25 0x00007ffff0c2c15e in QAbstractButton::mouseReleaseEvent (this=0x55555d6df560, e=0x7fffffffbcb0) at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qabstractbutton.cpp:976
#26 0x00007ffff0d5729d in QToolButton::mouseReleaseEvent (this=0x55555d6df560, e=0x7fffffffbcb0) at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qtoolbutton.cpp:588
#27 0x00007ffff0b51276 in QWidget::event (this=0x55555d6df560, event=0x7fffffffbcb0) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidget.cpp:8963
#28 0x00007ffff0afe31a in QApplicationPrivate::notify_helper (this=this@entry=0x5555556168e0, receiver=receiver@entry=0x55555d6df560, e=e@entry=0x7fffffffbcb0) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3294
#29 0x00007ffff0b028d1 in QApplication::notify (this=<optimized out>, receiver=<optimized out>, e=0x7fffffffbcb0) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:2772
#30 0x00007ffff5998f0a in Gui::GUIApplication::notify (this=0x7fffffffcc80, receiver=0x55555d6df560, event=0x7fffffffbcb0) at /home/pieter/current/fpa/freecad-main/src/Gui/GuiApplication.cpp:93
#31 0x00007fffefb585a8 in QCoreApplication::notifyInternal2 (receiver=0x55555d6df560, event=0x7fffffffbcb0) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1165
#32 0x00007fffefb585fd in QCoreApplication::sendSpontaneousEvent (receiver=<optimized out>, event=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1623
[... 40 lines cut ...]

```shell
OS: Arch Linux (xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git)
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * OpenTheme 2024.5.16
  * osh-autodoc-workbench 0.2.3
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.29
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pieterhijma, 2024-10-28:
I'm not in a position to check, but my guess is that this is also the case in 0.21.2, so it can probably be regarded as something that doesn't occur often.

wwmayer, 2024-10-28:
I cannot confirm the crash.

What do you mean with?

 > Choose a plane in the 3D view

Which plane and how to select it? While the sketch is in edit mode I cannot select external geometries in the 3D view.

pieterhijma, 2024-10-28:
In step 3 "move to part design and create a sketch", If you press the button "Create Sketch", then a task opens in which you are asked to select one of the planes, for example the XY plane. You can also select a plane in the task panel under "Select Attachment". Do not do this at this point, but go to step 4. Then in step 5, you can select the XY plane in the 3D plane or in "Select Attachment".

wwmayer, 2024-10-28:
Now I got it. Thanks for the clarification!

wwmayer, 2024-10-28:
The crash already occurs at the deletion of the sketch.

When you click the undo button twice then the transaction that has created the sketch will be moved to the redo stack. If you now try to delete the remaining sketch then it first clears the redo stack where it already deletes the sketch and the body object. But as next step the system tries to nullify links of the sketch to other objects and this already causes undefined behaviour.

PR #17632 (2024-11-04): Fixes #17553
* Tests: Add unit test to demonstrate failure of #17553
* Fixes #17553
* If needed open another transaction for sketch creation
commit 400d5c49 (2024-11-04): Core: Fixes #17553
commit 8ac217a7 (2024-11-04): Merge pull request #17632 from wwmayer/issue_17553

Fixes #17553
commit f2bf2a4b (2024-11-04): Core: Fixes #17553
closed 2024-11-04 by yorikvanhavre