FreeCAD/FreeCAD #17554: Assembly not aligning properly for fixed joints

Buffalo1John, 2024-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a model of an axle and a train wheel. The assembly is to be one axle and two wheels. The axle was built from the center out to each end, so for example, the original pad was in both directions. It was done this way because of the symetrical nature of the drawing. The axle has groves in it where snap rings are to be placed, so I can use the edge of the groove to align to the edge of the wheel. The axle is "grounded" and one of the wheels has been rotated to face the right way. The axle has four sketches that make the grooves with pockets. The ends of the axle each have an additional two pockets, one on the outside of the end and the other on the inside of the end. There is a groove for both sides of each wheel. Here is the assembly just after the three bodies are added:
[image]

Now to create the fixed joint. First edge is on the axle:
[image]

Second edge is on the wheel and once selected, the wheel moves to the wrong place:
[image]

OK, so cancel that and lets try it from the other side of the wheel - first the axle groove edge:
[image]

And now the wheel edge:
[image]

OK, different, but also wrong, so cancel the joint. 

Now I see something, as I move fromone groove edge to another, the "joint axis" thingy (I don't know what it is called) is never lining up with the edge selected:
[image]
[image]
[image]
[image]

The other end of the axle is just as funky, but it actually has one edge that lines up properly (the last on of this series)
[image]
[image]
[image]
[image]

The Wheel, Axle, and Assembly files are zipped together and attached below.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * lattice2 1.0.0
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Anything else?

[WheelAxle.zip](https://github.com/user-attachments/files/17549408/WheelAxle.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Buffalo1John, 2024-10-28:
I messed up the last of the four pastes in the left side, the missing one is this:
[image]

maxwxyz, 2024-10-29:
@PaddleStroke could you take a look at this? Is it just displaying JCS at the wrong location?

MiqCG, 2024-10-29:
I think I have found that the problem is in the tree of the axle file: the Tip (small green icon) is set in the last operation of the tree but it is not activated.

@Buffalo1John , try to open the Axle file, set the visibility of the part at the last operation ("BotEndThreadPocket"), save the file, recompute the assembly and try to attack a fixed joint at the end of the axle. 

[image]

Tested with:
`OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * fasteners 0.5.29
`

Buffalo1John, 2024-10-29:
I can verify that hiding the "Tip" makes the joint fail and making the "Tip" visible makes the joint work correctly.

Saving the file does not seem to be neccessary for the success of the joint. 

If I make the Axle "Tip" visible, I can make the joint. If I delete that joint, hide the Axle "Tip" and try to make the joint, it will again fail. Curiously, if I make one of the wheels be the Grounded and hide the last pocket on the wheel, it makes joints properly. 

Thanks for identifying this, but it seems to me this is still a bug, I hide features of bodies in assemblies at times when I need to be able to see what I am doing. Since it lets me hide it, I use it that way...

maxwxyz, 2024-10-30:
The body always has the shape of the tip. So not sure if there is a bug here.

MiqCG, 2024-10-30:
> The body always has the shape of the tip.
I think here is the point of Buffalo1John, because the tip was already at the last operation in the Axle file, it was just not visible.

maxwxyz, 2024-10-30:
But when the visible feature is manually changes, there is nothing to do. How would FreeCAD know what you want to display then.
So is there an actual bug here, otherwise I'll close the issue.

Buffalo1John, 2024-10-30:
In my simplistic view of the situation, making the tip hidden should not make selection of an edge, further up the tree, fail.

I'm not trying to select a edge modified by the pocket step which is the hidden tip, I am selecting a edge on a pocket that was 1 to 7 places up the tree from the tip.

Looking at this differently, if I go to Part Design and hide a bunch of items:
[image]

I can select the face of one of those pockets, make a sketch and pad it, and I get this:
[image]

All that I had hidden is now visible, which is expected. The Sketch and pocket were where I selected, unlike how Assembly behaves. Yes, I know placing a joint and placing a sketch are completely different things, but I would expect them to behave similarly.

maxwxyz, 2024-10-30:
I think this is TNP related. If you hide the last feature, the edge name changes, as the pocket (tip) adds edges to the model.
You can see this in the status bar or selection dialog:
[image]
[image]

So it is not an alignment problem, but the edge changes and the new edge is at a different location.

@PaddleStroke @CalligaroV you are the experts here, is this something to consider or not planned?

CalligaroV, 2024-11-04:
@maxwxyz, your analysis looks correct to me, most likely this is a TNP issue.

Can't say ATM if the issue is in TNP code or in the recent TNP-Assembly PRs, I'll have to investigate.

Unfortunately ATM I can't tell you when I'll be able to do so, for now I'm following other issues.
But as said other times, if I don't give you updates don't hesitate to ping me again!

TIA!

Buffalo1John, 2025-08-21:
I am not sure how to see a daily build of a windows version, I just downloaded the weekly build:

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/20 00:00:55
Build type: Release
Branch: main
Hash: 5e4fe3a3c3af46be530969d323368fee8bfcba3b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * A2plus 0.4.68
  * Curves 0.6.61
  * fasteners 0.5.38
  * lattice2 1.0.0

still shows the problem - hide the tip (and a bunch more) of the axle 

[image]

and when trying to select for the fixed joint either of the two grooves, the location of the joint (the axis) is in the wrong place. Here are capture os two images pasted so you can see:

[image]

I can wait for the next weekly build and try that - or the original file I uploaded seems to have exceptions but I have a newer version that does load properly - tip is hidden, recompute done with no errors, shows problem...

[WheelAxleAssembly.zip](https://github.com/user-attachments/files/21927065/WheelAxleAssembly.zip)

luzpaz, 2025-11-26:
> I'll try the file here in a second

@drwho495 any progress ?

drwho495, 2026-06-06:
Changing the shown feature won't change the shape that the assembly workbench has access to. Changing the tip feature will. The issue here comes from the selection aspect of joint creation (at least from what I can tell).

How assembly joints store selection references is by using the shape of the partdesign body, NOT the shapes of the individual features in said body. That's why the shape seen by the assembly workbench only changes when the tip changes. Whenever the joint creation task is open, you select elements from the shown feature shape, and not the partdesign body's shape. So if the shown feature is NOT the tip, there can be a mismatch in selected Indexed Name (which is like Edge1, Face2 or Vertex3) from the correct Indexed Name in the tip shape.

This should be an easy fix in the python code.

Buffalo1John, 2026-08-30:
> > Also - wouldn't this not really be a toponaming bug since it is really a choice of feature recorded rather than the recorded name not updating correctly when model changes are made?
> > It is still a bug, not toponaming but a design choice...
> 
> Correct. If we used toponaming properly here then it would be fixed.

@PaddleStroke - any progress onfixing this one?

pushkin-ale, 2026-10-12:
I ran into what looks like the same root cause, with a hard exception instead of a misplaced joint. I'm adding details in case they help.

**Environment:** FreeCAD 26.3.0 (Libs 26.3.0RC1R20261008), macOS

**Symptom:** A joint on a Body linked into an Assembly throws on every recompute:

```
UtilsAssembly.py, line 1066, in findPlacement
    edge = get_element(obj.Shape, elt)
UtilsAssembly.py, line 1167, in get_element
    element = shape.getElement(name)
IndexError: Shape index 63 out of bound 44
```

**Setup:** The Body's feature chain is `... -> PolarPattern -> Mirrored`, but the Tip is on `PolarPattern`. `PolarPattern` has 44 edges and `Mirrored` has 86. The link in the Assembly still displays the `Mirrored` geometry, so I could select an edge on it.

**What I found:** Joint references are stored with a feature prefix (e.g. `Mirrored.Edge63`, `Fillet.Face37`). `findPlacement` resolves the index against `obj.Shape`, which for a Body is the Tip's shape. It ignores the prefix. I reproduced this by script on the affected model with the Tip on `PolarPattern`:

- `Mirrored.Edge63` raises `IndexError: Shape index 63 out of bound 44`
- `Edge63` fails the same way
- `PolarPattern.Edge20` resolves fine

So the Assembly displays and lets the user pick one shape (the visible feature), but computes the joint from another (the Tip). Out of range, it throws. In range, it silently picks the wrong edge, which matches the misplaced joint described here.

**Suggestion (UX):** If the Assembly only works with the Tip's geometry, it should show only the Tip's geometry. Rendering a different feature and letting the user pick elements from it is misleading. Nothing in the UI tells the user that Tip position and feature visibility affect joints, and re-creating the assembly doesn't help because the cause is inside the Body. Concretely:

1. When a Body is linked into an Assembly, render the Tip's shape, regardless of per-feature visibility, so what you see is what you can pick and what the solver uses.
2. If that isn't feasible, restrict picking to elements of the Tip's shape, or resolve the reference against the feature named in the prefix.
3. At minimum, report "referenced element is not on the Body's Tip" instead of a raw `IndexError`.

I filed this as #33428, which was closed as a duplicate of this issue.

Buffalo1John, 2026-10-12:
I tried it with the files I uploaded Bug17554.zip and I can't get my extremely simple example to throw an error. I did get a pile of:
"<TopoShape> TopoShapeExpansion.cpp(1000): hasher mismatch"
when I try to make a joint (fills and scrolls the Report View). Recompute is not an available function and making the joint, deleting that joint, and making a new joint gets same message.

PaxTux, 2026-10-13:
This issue and [https://github.com/FreeCAD/FreeCAD/issues/28538](https://github.com/FreeCAD/FreeCAD/issues/28538) are duplicates

luzpaz, 2026-10-14:
Any chance we can get this looked at before the upcoming 26.3 release ?