FreeCAD/FreeCAD #17556: Assembly workbench fails when making assemblies

picli3, 2024-10-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A few versions ago the assembly worked fine, however with the new updates it goes out of control with a simple joint.
[image]

As you can see in the video, it works fine at first, but then when I press OK, it goes out of control. Well, maybe you can think, "you have a parallel relationship with the upper face," but this doesn't work either.

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Peru (es_PE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * A2plus 0.4.66
  * kicadStepUpMod 11.2.9
  * ExplodedAssembly
```

### Subproject(s) affected?

Assembly

### Anything else?

[SA_box.zip](https://github.com/user-attachments/files/17556006/SA_box.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-29:
Can confirm it behaves weird. You can rotate it 90 degrees to be the same orientation again but trying to constrain it with a parallel joint to the bottom plane breaks the distance joint.

FEA-eng, 2024-10-29:
Shouldn't this be a blocker? Another pretty serious report regarding Assembly joints: https://forum.freecad.org/viewtopic.php?t=86451&start=160#p789434

maxwxyz, 2024-10-29:
at least for discussion

MiqCG, 2024-10-30:
I am not sure that it is having a bad behaviour. The 90 degrees fliping is due to the difference in orientation between the two joint markers (former JCS), so when the solving is done, they get aligned. With this flip, the constraint is not infringed. 

On the other hand, I think that the failing of the Parallel joint to align the "SA_sup" to the bottom edge of the grounded part is due to the 90 degrees but if you transform just a bit  the orientation of the SA_sup, the solver does the job. I have seen this limitation of not being able to completely rotate a body in various solvers.

[image]

MiqCG, 2024-10-30:
> But the parallel joint also breaks the distance joint somehow, can you reproduce this too?

Sorry, that wasn't my case: the Parallel joint doesn't brake the Distance joint. Only looks like it does because of the perspective.
I attach the files:
[SA_box_MiqCG.zip](https://github.com/user-attachments/files/17579395/SA_box_MiqCG.zip)

`
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * freecad.gears 1.3.0
  * fasteners 0.5.29
`

picli3, 2024-10-31:
> > Pero la articulación paralela también rompe la articulación a distancia de alguna manera, ¿puedes reproducir esto también?
> 
> Lo siento, ese no fue mi caso: la junta Paralela no frena la junta Distancia. Solo parece que lo hace por la perspectiva. Adjunto los archivos: [SA_box_MiqCG.zip](https://github.com/user-attachments/files/17579395/SA_box_MiqCG.zip)
> 
> ```
> OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
> Word size of FreeCAD: 64-bit
> Version: 1.1.0dev.39092 (Git) AppImage
> Build type: Release
> Branch: main
> Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
> Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
> Locale: Catalan/Spain (ca_ES)
> Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
> Installed mods: 
>   * freecad.gears 1.3.0
>   * fasteners 0.5.29
> ```

So you mean there's no problem?

MiqCG, 2024-10-31:
I mean I couldn't find a wrong behaviour on my end. Can you check if the files that I attached works correctly for you?

maxwxyz, 2024-10-31:
Yes your file works on my system to without issues.
Dropping the blocker tag.

picli3, 2024-10-31:
My distribution is Linux Mint Wilma, I will try another one, I will look for another computer with Ubuntu to see if the problem continues.

MiqCG, 2024-10-31:
Talking privately with @picli3 in Spanish, he has showed me another thing that is a little bit more problematic.

As seen until now, if you execute a Distance joint between the Grounded part and "SA_sup" part, the parts get together in the preview (before OK) and then the SA_sup rotates 90º as commented. The report view gets dozens of the same warning: <TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch

**But**, if before all of this, the "SA_sup" part is rotated 180º with the transform tool along the green axis (Y), and then the Distance joint is applied, the preview doesn't get the two parts in contact, although when OK is pressed, the result is the same (SA_sup rotated 90º from the original orientation). As a difference, apart of the aforementioned warnings, this 3 error appear: 
338.261 <Assembly> AssemblyObject.cpp(176): Solve failed: iterNo > iterMax
338.207 <Assembly> AssemblyObject.cpp(176): Solve failed: iterNo > iterMax
172.697 <Assembly> AssemblyObject.cpp(176): Solve failed: iterNo > iterMax

closed 2025-10-25 by maxwxyz