FreeCAD/FreeCAD #17558: Sketcher external geometry not updating

leho-l, 2024-10-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When changing parameters the model updates, but the corresponding sketcher external geometry doesn't update.
Model has ~86 faces. When updating the model parameters it allso sometimes  gets these TPN problem, where a face shapebinder has gone  walking.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) Snap 1176
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: Estonian/Estonia (et_EE)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * PieMenu 1.8.0 (Disabled)
  * Curves 0.6.47
  * Ondsel-Lens 2024.7.5.02
  * btl 0.9.9 (Disabled)
  * Freecad-Built-in-themes-beta 1.2.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Anything else?

Model has ~86 faces. When updating the model parameters it allso sometimes  gets these TPN problem, where a face shapebinder has gone  walking.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-29:
Can you share an example file and steps how to reproduce?

Pendelf, 2024-11-25:
https://github.com/user-attachments/assets/cdcc4527-37ef-45be-9f75-7b96bcca42d7

[shape_binder_problem - 251124.zip](https://github.com/user-attachments/files/17899809/shape_binder_problem.-.251124.zip)

Is this it? Removed the roundings and after that shape binder does not update

Pendelf, 2024-11-25:
That's probably how it is

https://github.com/user-attachments/assets/e1080dcb-4087-497f-a18a-2d44762cb3f7

leho-l, 2024-12-12:
been making production parts and still there are some features not  behaving properly. 
With this model I have had to fix the same kind of external geometry not updating.
last  error: 
[Klaviatuuripakk.zip](https://github.com/user-attachments/files/18109324/Klaviatuuripakk.zip)
When readjusting the round through hole pocket004/sketch005 the CAM clone body and
features get updated and it gets some TopShapeExpanson.cpp errors.