FreeCAD/FreeCAD #17560: missing dependency in BIM objects library. Linux snap 1.0 RC2

ffpedraza, 2024-10-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

python3-requests library seems to be missing in Linux snaps, without this lib, the parts library doesn't work.

steps to reproduce:
1. open BIM workbench
2. open Objects Library
3. errors appear in the log and no objects or libraries can be used

these are the error messages: 
21:45:08  No structure in cache. Please refresh.
21:45:08  Traceback (most recent call last):
21:45:08    File "/snap/freecad/1162/usr/Mod/BIM/bimcommands/BimLibrary.py", line 900, in writeOfflineLib
21:45:08      rootfiles = self.getOnlineContentsAPI(LIBRARYURL)
21:45:08    File "/snap/freecad/1162/usr/Mod/BIM/bimcommands/BimLibrary.py", line 811, in getOnlineContentsAPI
21:45:08      import requests
21:45:08    File "/snap/freecad/1162/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
21:45:08      return original_import(name, *args, **kwargs)
21:45:08  ModuleNotFoundError: No module named 'requests'
22:34:09  No structure in cache. Please refresh.
22:34:09  Traceback (most recent call last):
22:34:09    File "/snap/freecad/1162/usr/Mod/BIM/bimcommands/BimLibrary.py", line 900, in writeOfflineLib
22:34:09      rootfiles = self.getOnlineContentsAPI(LIBRARYURL)
22:34:09    File "/snap/freecad/1162/usr/Mod/BIM/bimcommands/BimLibrary.py", line 811, in getOnlineContentsAPI
22:34:09      import requests
22:34:09    File "/snap/freecad/1162/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
22:34:09      return original_import(name, *args, **kwargs)
22:34:09  ModuleNotFoundError: No module named 'requests'

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38641 +165 (Git) Snap 1162
Build type: Release
Branch: tag: 1.0rc2
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: Spanish/Spain (es_ES)
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-29:
@yorikvanhavre FYI

yorikvanhavre, 2024-11-07:
Maybe we reopen this if it needs to be fixed in the flatpak too? Is it under our control?

closed 2024-11-07 by yorikvanhavre

adrianinsaval, 2024-11-07:
yes, I'm on it: https://github.com/flathub/org.freecad.FreeCAD/pull/6