FreeCAD/FreeCAD #17563: Assembly: Rack and pinion joint not working

FEA-eng, 2024-10-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Forum discussion: https://forum.freecad.org/viewtopic.php?t=86451&start=150#p789411

Apparently, there's a regression with the Rack and pinion joint - it has no effect.

 File (needs the Gears wb to work properly):

[lens_slider.FCStd.zip](https://github.com/user-attachments/files/17561865/lens_slider.FCStd.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-29:
@PaddleStroke FYI

FEA-eng, 2024-10-29:
Since this joint seems to be completely broken, it might be a blocker.

MiqCG, 2024-10-30:
Hi,

Today I listened elsewhere that Gear WB has become problematic. I haven't tested if it's true or not but I thought that could be related to this issue. 
I have made a very simple test file with Part (attached), and the Rack and Pinion joint works. I can see the "seam" of the cylinder passing by.

[AssRackPinion_v1.FCStd.zip](https://github.com/user-attachments/files/17568509/AssRackPinion_v1.FCStd.zip)

Tested with:
`
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git) AppImage
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * fasteners 0.5.29
`

maxwxyz, 2024-10-30:
Rack and pinion works in general. Could be related to GearsWB.
@FEA-eng Dropping the `Blocker` tag

Simple Demo:
[image]

```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.12
  * Curves 0.6.50
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.25
  * Rocket 3.3.0
  * sheetmetal 0.5.3
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

FEA-eng, 2024-10-30:
> Simple Demo

Can you share this file? Might be helpful for further testing.

maxwxyz, 2024-10-30:
Did not save it...
There is one in https://github.com/FreeCAD/FreeCAD/issues/17563#issuecomment-2446182413

flerli, 2024-10-30:
The forum post was originally from me. I just checked the joint with parts from part design WB and the rack and pinion joint works perfectly. So it looks like it is caused by either the rack or the pinion or both created in GearsWB.

I attached the working example.
[rack_pinion_joint_test.FCStd.zip](https://github.com/user-attachments/files/17573816/rack_pinion_joint_test.FCStd.zip)

MiqCG, 2024-10-30:
@maxwxyz , I am curious about what is the protocol in these cases.
Do we have to make a ping to the Gears WB manteiner? Do we have to open an issue in the Gears github project?

maxwxyz, 2024-10-30:
yes please open an issue there. Hope Pierre can still look into it though

MiqCG, 2024-10-30:
> yes please open an issue there.

Done: 
https://github.com/looooo/freecad.gears/issues/180

MiqCG, 2024-10-30:
Maybe I was hasty opening the issue in the looooo github.
Now I have made a simple example and the joint seems to work, so maybe it's a problem of the original file or the followed steps.

Take a look at the movement in the screencast, beside the misplaced objects because of laziness. 

[Rack_Pinion_Gears.FCStd.zip](https://github.com/user-attachments/files/17575908/Rack_Pinion_Gears.FCStd.zip)

https://github.com/user-attachments/assets/91637e89-5f38-4705-9eb9-ae6267609bb7

giulio-alfieri, 2025-02-25:
Hi folks, I’m a complete newbie to FreeCAD and I’m struggling to make a rack and pinion joint. The issue is that the movement direction is inverted. @MiqCG in your example, the movement of the gear is inverted against the rack. The same goes for @flerli´s example—the movement is inverted. What am I missing?

MiqCG, 2025-02-26:
Dear @giulio-alfieri ,

I recommend to you to use the FreeCAD forum for help petitions, there is a section on [Help on using FreeCAD](https://forum.freecad.org/viewforum.php?f=3). Before opening a new thread, is preferable to search for already existing answers to your problem. 

Github is mainly dedicated to report bugs and detected problems in the expected functionality of FreeCAD. Adding to that, this is a closed issue, so it is less likely to receive any answer.

Flavelius, 2025-08-16:
```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.2.38641 +678 (Git) Snap 1634
Build type: Release
Branch: tag: 1.0.2
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, OCC 7.7.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.68
  * freecad.gears 1.3.0

```

luzpaz, 2025-08-16:
@Flavelius can you reproduce on 1.1dev ? Please present About info for the version you can reproduce it on, TIA

Flavelius, 2025-08-17:
[rp_basic_objects.zip](https://github.com/user-attachments/files/21820787/rp_basic_objects.zip)
i used the appimage downloaded from github to create this file. The appimage had issues saving preferences and loading files, so i couldn't install the gears workbench and so only created the setup with basic primitives.
The version i downloaded is:
```
OS: Ubuntu 24.04.3 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.57/1
```
in both versions (this and the one i replied earlier), the basic shapes setup initially looks as if behaving correctly. But dragging around the sliding or rotating element a few times it starts to print warnings about bad solves until at some point it breaks (for me atleast) and nothing moves anymore.

FC-FBXL5, 2025-10-22:
Still doesn't work properly.

```
OS: macOS 26.0.1
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 72/133/2
Installed mods: 
  * Reinforcement
  * Movie 2025.1.4
  * freecad.gears 1.3.0
  * Templater 0.0.2
  * AddonManager 2025.9.23
  * fasteners 0.5.43
  * Ship 2024.11.26
  * Cables 0.2.1
  * sheetmetal 0.7.55
  * MakerWorkbench 1.0.1
  * Curves 0.6.70
  * Assembly3 0.12.2
```

FC-FBXL5, 2025-11-05:
Usually a rack is guided by a slider joint and a slider joint can be created by selecting two straight edges.
Can you detect the error? 
2 straight edges supply enough information to create a cylindrical joint but which element is used to keep the moving part from rotating around the Z-axis? It seems to be a quite intelligent algorithm in the background that automatically detects other elements to prevent rotation.
Could it be that this algorithm is a bit unstable when it comes to turning LCSs around?

closed 2026-09-30 by maxwxyz