FreeCAD/FreeCAD #17568: Weekly build 39092 conda macOS arm64 py311 "damaged" won't run

crunchysteve, 2024-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Drag from disk image to 2020 M1 (ARM64) MacBook Air Applications folder, I get this dialog.

[image]

Version 1.0.0RC2 runs fine but wanted to try the weekly release to see if there are any improvements.

### Full version info

```shell
FreeCAD_weekly-builds-39092-conda-macOS-arm64-py311.dmg
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-30:
See https://github.com/FreeCAD/FreeCAD-Bundle/pull/293#issuecomment-2298014913 for now until the weekly builds are signed

closed 2024-10-30 by maxwxyz

crunchysteve, 2024-10-30:
That worked, thanks :-)