FreeCAD/FreeCAD #17571: SyntaxWarning error with BIM's example file

akagohary, 2024-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When i open any example file its fine except for the BIM example, there must be something wrong with the BIM suite

```
16:08:20  /usr/lib/freecad/Mod/BIM/ArchAxis.py:401: SyntaxWarning: invalid escape sequence '\['
  pts = re.findall("point \[(.*?)\]",buf)[0]
16:08:20  /usr/lib/freecad/Mod/BIM/ArchSite.py:70: SyntaxWarning: invalid escape sequence '\['
  buf = re.findall("point \[(.*?)\]",buf)
16:08:20  /usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:737: SyntaxWarning: invalid escape sequence '\#'
  marker = re.findall("<line x1=.*?stroke=\"\#ffffff\".*?\/>",svg)
16:08:20  /usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:753: SyntaxWarning: invalid escape sequence '\#'
  svg = re.sub("<line x1=.*?stroke=\"\#ffffff\".*?\/>","",svg,count=1)
16:08:20  /usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:781: SyntaxWarning: invalid escape sequence '\?'
  svg = re.sub("<\?xml.*?>","",svg,flags=re.MULTILINE|re.DOTALL)
16:08:20  /usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:783: SyntaxWarning: invalid escape sequence '\/'
  svg = re.sub("<\/svg>","",svg,flags=re.MULTILINE|re.DOTALL)
```

Steps to reproduce :
1- Open FreeCAD ver 1.0.0RC2.38806 
2- Click on BIMExample
3- Log with errors pops up

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-30:
@yorikvanhavre FYI

luzpaz, 2024-10-30:
Can't reproduce on  
```
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

Syres916, 2024-10-30:
@luzpaz it's only triggered by Python 3.12+ is my understanding.

luzpaz, 2024-10-30:
@cjmayo do you mind commenting in this ticket so I can assign this ticket to you? TIA

hoshengwei, 2024-11-02:
Can't reproduce on  python 3.12 also
```
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.39100 +8 (Git)
Build type: Release
Branch: StorySettingManager
Hash: 46880114009d10d8f4a82b021e533383463fae83
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
```

akagohary, 2024-11-03:
> Can't reproduce on python 3.12 also
> 
> ```
> OS: Arch Linux (GNOME/gnome/wayland)
> Architecture: x86_64
> Version: 1.1.0dev.39100 +8 (Git)
> Build type: Release
> Branch: StorySettingManager
> Hash: 46880114009d10d8f4a82b021e533383463fae83
> Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
> Locale: English/United States (en_US)
> Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
> ```
> 
> maybe try install ifcopenshell first?

i  installed ifcopenshell locally in site packages and tried again as per the wiki entry ["Using a pre-compiled IfcOpenShell package"](https://wiki.freecad.org/IfcOpenShell) 

[image]

```
Python 3.12.7 (main, Oct  1 2024, 11:15:50) [GCC 14.2.1 20240910] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.runCommand('Std_Workbench',10)
>>> # Gui.runCommand('Std_ViewStatusBar',1)
>>> # Gui.runCommand('Start_Start',0)
>>> import ifcopenshell
/usr/lib/python3.12/site-packages/ifcopenshell/file.py:412: SyntaxWarning: invalid escape sequence '\d'
  reg = "".join(f"(?P<{s}>{s}\d+)?" for s in prefixes)
>>> 
No stream support: No module named 'lark'
>>> print(ifcopenshell.version)
0.8.0
>>> print(ifcopenshell.__path__)
['/usr/lib/python3.12/site-packages/ifcopenshell']
```

i still get this although the `<class 'ModuleNotFoundError'>: No module named 'ifcopenshell'` did go away

```
07:27:18  /usr/lib/freecad/Mod/BIM/ArchAxis.py:401: SyntaxWarning: invalid escape sequence '\['
  pts = re.findall("point \[(.*?)\]",buf)[0]
07:27:18  /usr/lib/freecad/Mod/BIM/ArchSite.py:70: SyntaxWarning: invalid escape sequence '\['
  buf = re.findall("point \[(.*?)\]",buf)
07:27:18  /usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:737: SyntaxWarning: invalid escape sequence '\#'
  marker = re.findall("<line x1=.*?stroke=\"\#ffffff\".*?\/>",svg)
07:27:18  /usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:753: SyntaxWarning: invalid escape sequence '\#'
  svg = re.sub("<line x1=.*?stroke=\"\#ffffff\".*?\/>","",svg,count=1)
07:27:18  /usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:781: SyntaxWarning: invalid escape sequence '\?'
  svg = re.sub("<\?xml.*?>","",svg,flags=re.MULTILINE|re.DOTALL)
07:27:18  /usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:783: SyntaxWarning: invalid escape sequence '\/'
  svg = re.sub("<\/svg>","",svg,flags=re.MULTILINE|re.DOTALL)

```

hoshengwei, 2024-11-03:
https://github.com/FreeCAD/FreeCAD/pull/16857

this bug might already fixed around 1 month ago
I tried compile myself and using the AppImage
it all works without error

but I did reproduce this error by using `pacman -S freecad` to install FreeCAD

I think this is not the FreeCAD problem, more like Pacman not update to using the up-to-date source
maybe you could try using the AppImage Version first?

hoshengwei, 2024-11-03:
I think you misunderstanding what I mean

1.  I can reproduce this bug   only  if I install FreeCAD via Pacman  
2.  Arter checking,   found out this `re SyntaxWarning` bug is fix in PR https://github.com/FreeCAD/FreeCAD/pull/16857 already,    you better check the link and commit
3.  I think the problem now is not from FreeCAD but Pacman,  so maybe you need to wait pacman update the source code
4. I know the appimage is py311,  what I mean is if you want to using FreeCAD fully functional, just try using the AppImage not pacman.   
5. I think we can close this issue now.

akagohary, 2024-11-04:
No i did understand what you're saying correctly, but what i need to confirm is that the release am on `Version: 1.0.0RC2.38806` doesn't include the merge you mentioned so we can confirm that its the version on archlinux, cause all the other workbenches don't give me those errors its just BIM workbench, the package on archlinux is made on 14th of October, and the merge was on 30th of September, that means if the package was made from the main branch it shouldn've had it. And already a pull request is made on this issue with some changes so idk why would you discard that without even commenting on what was done in that pull https://github.com/FreeCAD/FreeCAD/pull/17583

hoshengwei, 2024-11-04:
sorry , what I mean the problem is not pacman itself,  its the FreeCAD package it used
you can check your error report
and go your system 
to check the code in
```
/usr/lib/freecad/Mod/BIM/ArchAxis.py:401:
/usr/lib/freecad/Mod/BIM/ArchSite.py:70:
/usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:737:
.....
```
compare the source code in the github
the problem is solved already in the PR #16857 

the total commit for now is 39101
if you want to check ver 38806
39101-38806 = 295
you might check the commit history for previous 295 commits is including that PR or not
I think 38806 using the commit  around Sep 23

I am not sure what #17583 is trying to fix.
maybe there are still some escape sequence not been caught in #16857 
but I am sure that fix is nothing to do with your warning report
like `src/Mod/BIM/importers/importIFClegacy.py`  it commit
the BIM Ifc import is already not using this module anymore

maybe that PR is fix somthing
but I think that is nothing about your error report
your error is just pacman package problem not FreeCAD itself

akagohary, 2024-11-04:
> sorry , what I mean the problem is not pacman itself, its the FreeCAD package it used you can check your error report and go your system to check the code in
>
> ``` [...]

Thank you so much for explaining the commit number/build number relation, i'll try and check when the commit was made precisely, i think its worth a shot to try and build freecad from the main branch and try it out especially with new commits being made in lesser frequencies, if i did build freecad and the problem didn't show up i'll post here again with my feedback, although am not sure when am gonna be able to try it.

Thank you again for your time :).

akagohary, 2024-11-04:
@hoshengwei i tried to compile from source, and i was on dependency hunt and everything seemed to be going, untill i hit his error which i can't fix to test the program.

`<class 'RuntimeError'>: No SWIG wrapped library loaded`

for alottttt of modules

although i have SWIG from arch repo installed, and my cmake detected swig for the build, i searched but couldn't find anyone with the same problem.

akagohary, 2024-11-04:
Nevermind i just had to recompile after maybe missing a dependency and it had to be there idk during cmake or make, but rebuilding solved my problem.

Also syntax error is no longer happening its fixed and indeed the package in Arch Linux is just outdated.

I will close this issue " idk if its the correct thing to do or not if not please reopen ", since the issue was afterall not related to FreeCAD and indeed to the Arch Linux package.

Thank you so much @hoshengwei for the insight.

akagohary, 2024-11-04:
@maxwxyz

akagohary, 2024-11-04:
ok this is ridiculous lol, when i opened assembly example

```
14:53:51  /usr/lib/python3.12/site-packages/ifcopenshell/file.py:412: SyntaxWarning: invalid escape sequence '\d'
  reg = "".join(f"(?P<{s}>{s}\d+)?" for s in prefixes)
```

```
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.1.0dev.39101 (Git)
Build type: Release
Branch: main
Hash: b7bdf93654c3940645d8c224bb7b24cf590cb79d
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

```

yorikvanhavre, 2024-11-05:
> 14:53:51  /usr/lib/python3.12/site-packages/ifcopenshell/file.py:412: SyntaxWarning: invalid escape sequence '\d'
  reg = "".join(f"(?P<{s}>{s}\d+)?" for s in prefixes)

This is a bug in ifcopenshell... Starting from python 3.12 all regex strings must become raw strings. Maybe your version of IfcOpenShell needs to be updated, where did you get it from?

akagohary, 2024-11-05:
> This is a bug in ifcopenshell... Starting from python 3.12 all regex strings must become raw strings. Maybe your version of IfcOpenShell needs to be updated, where did you get it from?

from here https://docs.ifcopenshell.org/ifcopenshell-python/installation.html

zip packages and copied it to /usr/lib/python3.12/site-packages since archlinux doesn't have the package

PR #17583 (2024-11-05): Resolve Python invalid escape sequence warnings
Closes #17571.