FreeCAD/FreeCAD #17572: Using the Measure tool crashes FreeCAD...

CIKoolK, 2024-10-30:
[edit 10/30/2024 more testing at bottom]

I just found last evening that using the Measure tool crashes the last 3 or 4 builds on my Linux Mint v22 system (the last build that works is 38946:

1. Launch FC (I just used build 39092), open a new empty file;
2. Create a body;
3. Add a simple object (like a box;
4. Open the Measure tool;
5. Attempt to measure the length of any edge of the box;
6. FC crashes;

The last build that works is 38946, 1.0.0 RC2 (build 38806) works too.

Here's part of what I Get running it in a terminal session (I tried posting the entire thing is  64 lines (I.e A great TL;DR candidate):

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7c5151845320]
FreeCAD/FreeCAD-Bundle#1  0x7c5154dc2583 in Gui::DAG::Model::slotChangeObject(Gui::ViewProviderDocumentObject const&, App::Property const&) from /media/Applications/FreeCAD-39092/usr/bin/../lib/libFreeCADGui.so+0x163
FreeCAD/FreeCAD-Bundle#2  /media/Applications/FreeCAD-39092/usr/bin/../lib/libFreeCADGui.so(+0x50217c) [0x7c5154b0217c]
FreeCAD/FreeCAD-Bundle#3  0x7c5154b05517 in Gui::Document::slotChangedObject(App::DocumentObject const&, App::Property const&) from /media/Applications/FreeCAD-39092/usr/bin/../lib/libFreeCADGui.so+0x207
FreeCAD/FreeCAD-Bundle#4  /media/Applications/FreeCAD-39092/usr/bin/../lib/libFreeCADApp.so(+0x22a514) [0x7c5153a2a514]
FreeCAD/FreeCAD-Bundle#5  0x7c5153a65ef4 in App::DocumentObject::onChanged(App::Property const*) from /media/Applications/FreeCAD-39092/usr/bin/../lib/libFreeCADApp.so+0x134
FreeCAD/FreeCAD-Bundle#6  0x7c5153b83a65 in App::Property::hasSetValue() from /media/Applications/FreeCAD-39092/usr/bin/../lib/libFreeCADApp.so+0x135
FreeCAD/FreeCAD-Bundle#7  0x7c5153bf4da8 in App::PropertyString::setValue(char const*) from /media/Applications/FreeCAD-39092/usr/bin/../lib/libFreeCADApp.so+0x138
FreeCAD/FreeCAD-Bundle#8  0x7c513b5dac37 in MeasureGui::ViewProviderMeasureBase::updateData(App::Property const*) from /media/Applications/FreeCAD-39092/usr/lib/MeasureGui.so+0x547
FreeCAD/FreeCAD-Bundle#9  0x7c5154f879a1 in Gui::ViewProviderDocumentObject::updateView() from /media/Applications/FreeCAD-39092/usr/bin/../lib/libFreeCADGui.so+0x101
FreeCAD/FreeCAD-Bundle#10  0x7c5154b06465 in Gui::Document::slotNewObject(App::DocumentObject const&) from /media/Applications/FreeCAD-39092/usr/bin/../lib/libFreeCADGui.so

My system:

```
OS: Linux Mint wilma 22 x86_64
Kernel: Linux 6.8.0-47-generic
Uptime: 21 mins
Packages: 2902 (dpkg)
Shell: bash 5.2.21
Display (100095550): 1920x1080 @ 75 Hz in 32" [External] *
Display (G276HL): 1920x1080 @ 60 Hz in 27" [External]
DE: Mate 1.26.2
WM: Marco (X11)
WM Theme: WinMeCIK
Theme: gtk2 [Qt], Adwaita [GTK2/3/4]
Icons: Mint-Y-Sand [Qt], Yaru-blue-dark [GTK2/3/4]
Font: qt5ct [Qt], Ubuntu (12pt) [GTK2/3/4]
Cursor: Take-1-White (48px)
Terminal: guake
CPU: AMD FX(tm)-8350 (8) @ 4.00 GHz
GPU: NVIDIA GeForce GT 740
Memory: 2.73 GiB / 31.32 GiB (9%)
Swap: 0 B / 2.00 GiB (0%)
Disk (/): 90.17 GiB / 509.13 GiB (18%) - ext4
Disk (/media/Applications): 23.31 GiB / 427.13 GiB (5%) - ext4
Disk (/media/Hitachi3TB): 1.26 TiB / 2.69 TiB (47%) - ext4
Disk (/media/TShift-2024): 69.73 GiB / 937.82 GiB (7%) - ext4
Disk (/media/[user]/Ventoy): 31.84 GiB / 460.22 GiB (7%) - exfat [External]
Local IP (enp5s0): 192.168.1.49/24
Locale: en_US.UTF-8

```

I'm just getting ready to boot Win 10 Pro (which I REALLY do not like doing) to see how it goes there...

Did it:
The newest available  build for Windows was 39057, it worked as it should on Win 10 Pro--it (b39057) crashes as described above on my Linux system.

Damn, Windows is slow,,,

More Info:
Here's what I have found since the above:

- LMDE v6.0 Cinnamon: b39092 = crash;
- Mint v21.3 Cinnamon: b39092=  OK;
- Mint v22.0 XFCE: b939092 =  OK;
- Mint v22 Cinnamon: b39092 =  crash;

yorikvanhavre, 2024-10-30:
Can't reproduce either.. I can follow the procedure described above, no crash, and I get the desired result (the edge gets measured).
```
OS: Debian GNU/Linux trixie/sid (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Unknown
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.12.6, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * dxf-library
  * OpenTheme 2024.7.3
  * parts_library
  * freecad.gears 1.3.0
  * OpenDark 2024.3.13
  * CADExchanger
  * BIM021 2021.12.0
  * ExtremeProDark 2.9.0
```

wwmayer, 2024-10-30:
> FreeCAD/FreeCAD-Bundle#1  0x7c5154dc2583 in Gui::DAG::Model::slotChangeObject

That's not related to the measurement but the DAG view. For the time being you should disable it. In the parameter editor go to BaseApp > Preferences > DockWindows > DAGView and set the parameter Enabled to false.

PR #17581 (2024-11-04): DAG: Add method hasRecord() and use it in Model::slotChangeObject
to avoid a possible failing assert().

Fixes #17572
commit 4143c551 (2024-11-04): DAG: Add method hasRecord() and use it in Model::slotChangeObject

to avoid a possible failing assert().

Fixes #17572
commit 7a484438 (2024-11-04): DAG: Add method hasRecord() and use it in Model::slotChangeObject

to avoid a possible failing assert().

Fixes #17572
closed 2024-11-04 by yorikvanhavre