FreeCAD/FreeCAD #17574: UI/UX: Attachement editor Improvement

leoheck, 2024-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Attachment editor is messy.

The green label is not necessary. The current `Attachment mode:` should be selected when editing an existing LCS.
Sometimes it is not selected for some unknown reason.
[image]

Modes that cannot be selected immediately could have a different color..
In this case, the 2 items at the end...
[image]

The text could be improved too:
- the group title from `Attachement` to `Attachement Editor`
- from `Attachment mode:` to `Mode` since it is in the Attachment Editor.
- from `Attachment Offset (in local coordinates)` to `Offset (in local coordinates)`

These buttons here are horrible to use...
Dont we have a better solution.
[image]
Suggestion, start by setting a constant length because these things are always dancing... 

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-30:
@FreeCAD/design-working-group FYI

pure-machinery, 2024-10-30:
I agree with the constant width of the buttons. Another solution is to just have a "pick icon" (same width always) and when it is actived it changes "state" or style until you pick a certain object.

Basically when you click on the icon it goes to the select mode for that attachment row. If you select something else the label next to is should be changed.  

Generally the picking of stuff in FreeCAD should somehow be improved. I never know when i'm in select mode or not and it's kind of quirky.

Additional suggestion is to change the mouse icon to the aforementioned "pick icon" - just another queue to the user that he is in select mode.

leoheck, 2024-10-31:
Definitely, a reset button per field is a must and probably a reset that cleans all the options at once too.

kaiwas, 2024-10-31:
> Definitely, a reset button per field is a must and probably a reset that cleans all the options at once too.

Yes. That would be great. (I updated my picture in the previous post). Thanks for the idea!

pierreporte, 2024-10-31:
In my opinion, the current attachment panel has a good potential, especially the list of available attachment modes. The hardest part to use is actually understanding what kind of references you can have for what you want to attach.

What would be good is to limit available options to what is necessary to attach something, and maybe propose predefined options to guide the user. For example, a plane can be defined in different ways: offset from an existing plane, normal to a curve through a point, two co-planar lines, etc. When the user selects a plane as a first reference, the next options would be limited to what is compatible with it. A combo box so that there is a limit in place from the start would help users to determine the best action depending on what they want to achieve. The attachment modes are very useful here because when a point is needed as a reference, users can select implicit points (e.g. center of a circle). I actually proposed something similar already for sketches in https://github.com/FreeCAD/FreeCAD/issues/12106#issuecomment-2106176656.

For points, it should be easier to enter arbitrary coordinates.

Overall, quality of life features already discussed here would be welcome.

pure-machinery, 2024-10-31:
Exaclty! It would be pretty dope to have to pre select the attachment mode and then pick the attachments. The user can either be  given a maximum amount of options for selected attachment and dynamically "lock" reference additional fields that would over define the attachment.

Here's an amateur take on it :) 

[image]

pierreporte, 2025-01-19:
I’m copying the proposal discussed within the @FreeCAD/design-working-group 

------------------

I proposed to redesign the attachment panel of sketches but I also think that it could be a good idea to redesign all attachment panels. Currently there is only one for everything. Basically I want to have a different one for each kind of things to attach: sketches, datum points, datum lines, datum planes, and datum LCS.

Inspired by other CAD programs, I see something like this:

* First thing is selection of attachment mode (currently it’s second), filtered for relevant geometry (so a point will have a different list from sketches, planes, lines, and LCSs).
* Second, the reference input widgets, that change depending on the attachment mode selection: the number of inputs and their label corresponds to the selected mode.
* Third the necessary options for each mode (like invert direction, things like that)
* Four, optional offsets as today but with a tweak: now it is defined by global coordinate system, but I propose a local coordinate system as well based on the coordinates system of the un-offset geometry

It’s quite close to Catia features (except it doesn’t have offsets):

* http://catiadoc.free.fr/online/cfyug_C2/cfyugaxis.htm
* http://catiadoc.free.fr/online/cfyug_C2/cfyugaxissystem.htm
* http://catiadoc.free.fr/online/cfyug_C2/cfyugplane3D.htm
* http://catiadoc.free.fr/online/cfyug_C2/cfyugpoint3D.htm
* http://catiadoc.free.fr/online/cfyug_C2/cfyugline3D.htm

Big difference with Catia is that I think we shouldn’t allow mode to change with selection when the panel is active because it’s sometimes frustrating when working, but the most relevant mode could be selected and input filled with user selection prior feature activation.

We could think of new attachment modes that are missing (e. g. points by coordinates).

The two pictures below are mockups I did for a new sketch attachment a while back (so it doesn’t have local offsets) and a generic one for datum so you can see how it should look on principle without doing all the modes and options. 

[image]

[image]