FreeCAD/FreeCAD #17576: FreeCAD freezes if lockfile already exists

adrianinsaval, 2024-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

see https://github.com/flathub/org.freecad.FreeCAD/issues/4 and https://github.com/flathub/org.freecad.FreeCAD/issues/5 for whatever reason when running as flatpak `QCoreApplication::applicationPid()` always returns 2 so if you start a second instance of freecad it tries to create an existing lockfile and freezes up. There could be two somewhat separate problems here:

* freecad probably shouldn't freeze because a lockfile exists, in any case it should raise an error? not sure
* using PID is clearly not unique enough in flatpak environment and possibly other situations, maybe we should generate an UUID early in startup and store it for later use instead of the PID. I found 3 uses of `QCoreApplication::applicationPid()`, I don't know if we use PID somewhere else through a different function.
https://github.com/FreeCAD/FreeCAD/blob/b6cda4309a9031bebffa73b694e9a9cc125d64de/src/App/Document.cpp#L944
https://github.com/FreeCAD/FreeCAD/blob/b6cda4309a9031bebffa73b694e9a9cc125d64de/src/Gui/Application.cpp#L2177
https://github.com/FreeCAD/FreeCAD/blob/b6cda4309a9031bebffa73b694e9a9cc125d64de/src/Gui/DocumentRecovery.cpp#L657

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 3d63fc6)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-10-30:
@wwmayer @chennes would appreciate opinions

adrianinsaval, 2024-10-30:
could there be value in using qlockfile instead of boost?

wwmayer, 2024-10-31:
boost's `file_lock` has the method `try_lock()` that returns a boolean. If acquiring the ownership of the file fails it returns false and true otherwise.

xtemp09, 2024-10-31:
_Qt_ also seems to have `bool QLockFile::tryLock(int timeout)`.[^1]

[^1]: https://doc.qt.io/qt-6/qlockfile.html#tryLock

adrianinsaval, 2024-10-31:
> _Qt_ also seems to have `bool QLockFile::tryLock(int timeout)`.[^1]
> 
> 
> [^1]: https://doc.qt.io/qt-6/qlockfile.html#tryLock

Introduced in 6.2, our minimum supported version is 5.12 I believe

xtemp09, 2024-11-01:
@adrianinsaval, how about [this one](https://doc.qt.io/qt-5/qlockfile.html#tryLock)?

chennes, 2024-11-01:
That should be sufficient -- we will not ever support Qt 6.0 or 6.1, so their lack of that method should not affect us.

wwmayer, 2024-11-01:
I have tried the QLockFile class and it doesn't work as reliably as boost's `interprocess::file_lock`.

When using QLockFile in tryRunEventLoop then it should also be used in DocumentRecoveryHandler::checkForPreviousCrashes.

If a previous FreeCAD session has crashed it leaves its lock file and possibly some directories. In a subsequent FreeCAD session QLockFile should be able to lock the old files but here in 70-80% of all cases it failed. I had to run FreeCAD 4-5 x until QLockFile was able to lock the file.

xtemp09, 2024-11-01:
What about `QSystemSemaphore`? [^1]

[^1]: https://evileg.com/en/post/147/

wwmayer, 2024-11-01:
What's wrong with boost's `interprocess::file_lock`? As said before this class also provides a method `try_lock` that returns immediately without blocking the application.

wwmayer, 2024-11-01:
There is now a tiny wrapper function `Application::applicationPid()` for Qt's applicationPid(). For now it simply calls `QCoreApplication::applicationPid()` but if needed it can be changed to return a (static) random number.

adrianinsaval, 2024-11-01:
> but if needed it can be changed to return a (static) random number.

that would be helpful for the document restore problem on flatpak I think

wwmayer, 2024-11-02:
> that would be helpful for the document restore problem on flatpak I think

That would mainly be helpful to allow multiple FreeCAD instances.

closed 2024-11-04 by yorikvanhavre

adrianinsaval, 2024-11-05:
created #17678 for the remaining problem

PR #17609 (2024-11-04): Core: Don't freeze application if lock file already exists
Fixes #17576