FreeCAD/FreeCAD #17577: Placement (Rotation) flips when part is modified

moedn, 2024-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hey there! We're using FreeCAD to build an open-source MRI scanner (at least as far es possible for the mechanical parts); one of the parts are so called "shim trays" = trays holding permanent magnets to passively balance out tolerances of a magnetic field lying between those trays.
The FreeCAD file contains 12 trays – 11 copies from `magnet-insert 1` as a rotational pattern. Most features – and also the Placement of the tray – are defined using Spreadsheets, see screenshot below. When I want to cut the magnet Pockets (using a FreeCAD script, sub-objects shape binder and the Pocket operation (see steps-to-reproduce below), the rotation angle in Placement flips from positive to negative, or vice versa. That the case for the other parts as well and also for other users on other systems. My workaround for now is to add an extra `-` into the formula _after_ I'm done cutting the pockets, so the part comes back into place – performed for each part at a time.

[image]

- link to file: https://gitlab.com/osii/shimming/shim-trays/-/blob/rc/2.1/src/magnet-insert.FCStd
- steps to reproduce: https://gitlab.com/osii/shimming/shim-trays/-/tree/rc/2.1/src?ref_type=heads#making-the-magnet-pockets (however, if you want to use the sample data: it's not at the right radius yet; you can either adjust `r_shim` in the spreadsheets accordingly or try it on rc/1.2 where the issue came up as well)

### Full version info

```shell
v0.21.2.33771
```

### Subproject(s) affected?

None

### Anything else?

just a gut feeling, but maybe this is related to #16983

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-30:
Please test if the issue is still present in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and post full about info. TIA

luzpaz, 2024-10-30:
An aside, @moedn please introduce yourself on our community forum.freecad.org to your project and team. We'd love to learn more details. Perhaps the community can also assist in your learning curve leveraging FC for your project?

kaiwas, 2024-10-30:
It's high time to replace the logic with vectors with angular Euler.
Working with vectors is useful, but in reality it is required very rarely. But rotations separately relative to the axes are required much more often (they are more visual and convenient), but in the current version they are only partially inoperative.

I tried for a long time to come up with a justification for the current approach, but I could not come up with possible reasons why rotations in the program initially work on vectors. But the most important thing is that it seems to suit everyone.

moedn, 2024-11-01:
Hey all! Thank you so much for the blazingly fast replies!

- @maxwxyz I'll test the latest dev version and post an update here
- @luzpaz happy to present myself and the project in the FreeCAD forum :) In fact, I'm pretty happy with FreeCAD and get a lot of support from FreeCAD experts in the community. There's a couple of moments where FreeCAD still bugs me out (e.g. losing references after altering previous features and messing up the part; I believe this is known as the internal renaming issue), but I have a strong faith that many of those issues will be resolved in v1.0 and updates in the near future :)
- @kaiwas I don't know how rotations are currently done in FreeCAD, but having worked in robitics a while ago, I can also only strongly recommend the use of quaternions instead of Euler Angles. The latter is just subject to all kinds of issues in special cases (such as gimbal lock), a source of errors when working with different notations and is often very cumbersome to work with ('#rotation-matrices)

BTW someone in Chile, who's trying to replicate the scanner and ran into the same issue, pointed out that the flip might come from a pos/neg change in `Axis`. In fact, on the screenshot above you can see the `-1` in the 3rd Axis field; I guess this is then flipping to `1`, still have to try it out and check again

moedn, 2024-11-01:
@maxwxyz Update from testing with v1.0rc2:

- I tried it on the shim tray version from rc/1.2 (where the issue came up for the first time); though recompute due to newer version fails; the error message refers to details in a report, but I don't know where to look for this report
- on the shim tray version from rc/2.1 (as linked above) recomputing works, the macro runs, shape binder seems to work, but when wanting to cut the pockets I'm getting stuck at the error message below ↓

[image]

other than that, the new interface looks really nice!

luzpaz, 2024-11-01:
> There's a couple of moments where FreeCAD still bugs me out (e.g. losing references after altering previous features and messing up the part; I believe this is known as the internal renaming issue), [...]

@moedn Please try to reproduce those issue in the RC builds or weekly builds. If you can reproduce, feel free to search those issue in this bugtracker. And if they aren't already reported, please open tickets for them. Thanks!

bdieterm, 2024-11-03:
Hi @moedn!

The problem you experienced is because you specify only the angle of the placement property, and therefore it can happen that the rotation axis is flipped after a some computations, as in your case.

I recommend you to fully specify the whole placement via an expression. To enter an expression for the placement property, right click on "Placement" (in the left "Property" column) and then click on "Expression..." in the context menu.

Instead of the expression for the angle
`-360 / 12 * (Spreadsheet.tray_pos - 1)`
use the following expression for the whole placement:
`placement(vector(0; 0; 0); vector(0; 0; 1); -360 / 12 * (Spreadsheet.tray_pos - 1))`

This is `placement(base; axis; angle)`.
See https://wiki.freecad.org/Expressions for details.

moedn, 2024-11-04:
Thanks!! Following this logic, I "locked" the value `1` in under the 3rd axis (see screenshot above that shows a `-1` at this position) using a formula. I guess that would be easier to understand and modify for our collaborators than the Expression property (which is a valid approach nontheless)

I've tested it and it works :tada:, so we've hereby found a doable workaround and my issue is solved :ok_hand:

closed 2024-11-04 by moedn