FreeCAD/FreeCAD #17579: Sketcher Solver Issues with Distant Constraints flip direction

QuinFromCanada, 2024-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/user-attachments/assets/6187db1e-9a0a-49ee-9af7-f7b70a211e30

If you draw 2 circles and use a distance constraint between the 2, if you make the inner shapes dimensions larger then the outer shape the 2 will flip position (inner becomes outer).

This happens with more advance shapes as well such as:

[image]
_Figure 1: Other test cases._

In my testing this happens when the Distance constraint is used (see Figure 2), and not when the Vertical Distance constraint or the Horizontal distance constraint are used (see Figure 3).  When I went back and checked my other test case (Figure 1), I noticed that the corner that was flipping directions had 2 Distance constraint rather then 1 Vertical Distance and 1 Horizontal distance.  Changing them to Vertical/Horizontal constraints fixed the issue.

[image]
_Figure 2:  Flip issue happens when the Distance Tool is used to separate these 2 shapes._

[image]
_Figure 3: Flip issue does NOT happen when using a Horizontal Distance Constraint._

My working theory is that the dimension tool does not have a Vector, so changing the inner object too much it will pull the object to it, rather then push it away.

This issue also exists in 0.21.

[image]
_Figure 4: Test on v0.21_

I feel that this is a big issue because with the new Dimensions tool it is very intuitive to click on 2 parallels lines if you want to create an offset between the 2.  When you do this it will create a Distance constraint by default,  and since all 3 of these tools look identical on the sketch surface,  it can leave to a lot of issues to new users.  In fact, this is exactly what I did in Figure 1.

In addition, when the inner/outers flip it can cause issues in other sketches that use external geometry, I tested this as well and it does in fact flip the external geometry in other sketches making it look like a possible TNP issue.

Let me know if you need an more information.

PS.  Big thanks to the whole FreeCAD team.  1.0 feels very nice, you can tell that a lot of love went into the release. Keep up the great work! 

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * Assembly3 0.12.0
  * CfdOF 1.22.9
  * Curves 0.6.13
  * fasteners 0.4.66
  * fcgear 1.0.0
  * freecad.gears 1.0.0
  * FreeCAD_assembly3-master
  * Movie 2023.6.10
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-10-30:
The first thing you can try is to change the solver in the sketcher.
Example in the video.

https://github.com/user-attachments/assets/c1a4a8fa-702d-42bb-b5d2-322b1e9b2147

I suggested that the developers change the default solver, but my idea was not supported. Probably, for some unknown reason, they prefer to suffer with the flipping problem until the problem is solved by itself.

awabom, 2025-02-25:
I've had this happen several times now. Today I didn't check after I changed a dimension in a sketch, and another sketch that used that geometry as external construction geometry got its constraints flipped and made a 10,5 mm distance into 9,5 mm, ruining a 47 minute 3d print. 

I tried changing the solver as suggested by @kaiwas, but:

- When using just a single sketch, the LevenbergMarquardt solver works and the larger circle stays larger.
- When using two sketches, one with a circle, and the other with external geometry + additional circle - it still doesn't work. Changing the 'base circle' sketch flips the constraint on the dependent sketch.

Also, the setting seems to be application-wide - so a project that needs another solver can malfunction when editing on another freecad installation.

When changing the 'BaseCircleSketch' to a larger diameter (e.g. 30mm), the 'ThisBreaksWithBothSolvers' sketch gets the constraints flipped in this file (from 1.1.0dev)

[DistanceConstraintsFlip.zip](https://github.com/user-attachments/files/18974240/DistanceConstraintsFlip.zip)

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: Dark-modern.qss/Dark-modern/Fusion
Logical/physical DPI: 96/93.1131
Installed mods: 
  * Curves 0.6.58
  * fasteners 0.5.37
  * FreeCAD-themes 2025.1.7
  * OpenTheme 2024.9.1
  * Render 2024.12.15
  * sheetmetal 0.7.11

kaiwas, 2025-02-26:
My opinion is that to solve such problems it is necessary to create a new type of constraints (possibly additional/auxiliary) that will indicate the direction.
for example, so that it would be possible to indicate that point B is located to the RIGHT of point A.

QuinFromCanada, 2025-02-27:
@kaiwas I'm not a programmer but in math terms I think the distance constraints needs a vector so that it doesn't flip itself around.

For a horizontal or vertical line you could do this by adding a flag bit to specify which point is closer to the origin (I'd imagine that this might already happens for the horizontal and vertical distance constraints since making both points of a line at the same coordinates gives you an "invalid solution" error).  

However, this solution wouldn't work for distance constraints since a none horizontal or vertical line has  a lot more chances of both points being equal distance from the origin.  You'd almost need 2 flags one for which point is closer to the x axis and another for which point is closer to the y axis.

drwho495, 2025-02-27:
having constraints to show what is in what order is pretty time consuming, In other platforms, you don't need to do stuff like this. I am also having this issue on the LevenbergMarquardt solver

drwho495, 2025-02-28:
Ok so I had an idea, when a distance constraint is made the element that comes first from the origin is recorded, and when the dimension is solved, if that element comes second then it is moved then solved again, which would fix this issue.

FEA-eng, 2025-11-12:
Flipping is definitely still an issue