FreeCAD/FreeCAD #17580: [Feature] CLI: Safe Mode should output same message to CLI as the GUI

luzpaz, 2024-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently when invoking from CLI  via `FreeCAD_xxx-x86_64.AppImage --safe-mode`

The output only shows (no indication that we're in Safe mode)
```shell
FreeCAD 1.1.0, Libs: 1.1.0devR39092 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Migrating Start Workbench to Start command... done.
```
While the GUi shows:  
 
[image]

#### Expectation  

CLI should show the same message in stdout

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

Core

### Anything else?

Related to #6106

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-31:
@luzpaz do you mean that it should be like this?
```
FreeCAD 1.1.0, Libs: 1.1.0devR39092 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

FreeCAD is running in safe mode.
Safe mode temporarily disables your configurations and addons. Restart the application to exit safe mode.

Migrating Start Workbench to Start command... done.
```

luzpaz, 2024-11-01:
Perhaps emphasize 'Safe mode.' ?

> FreeCAD is running in SAFE MODE.

> FreeCAD is running in \_Safe Mode\_

hyarion, 2024-11-01:
```
FreeCAD is running in safe mode.
FreeCAD is running in Safe Mode.
FreeCAD is running in SAFE MODE.

FreeCAD is running in _safe mode_.
FreeCAD is running in _Safe Mode_.
FreeCAD is running in _SAFE MODE_.

FreeCAD is running in _safe_mode_.
FreeCAD is running in _Safe_Mode_.
FreeCAD is running in _SAFE_MODE_.
```

So many versions to pick from 😅

luzpaz, 2024-11-01:
> FreeCAD is running in _SAFE_MODE_.

Lets go with this one for now. If any grievances arise from the community then it can be cosmetically modified later.

JamesHodgkins, 2024-11-02:
Happy to pick this one up unless anyone else wants to?