FreeCAD/FreeCAD #17582: Boolean union operation fails for large structures, when using Part::Feature objects, but not when using equivalent Part::Cylinder objects

mtav, 2024-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

### Bug description
I created a script to create _"Rod-Connected Diamond"_ unit-cells, which are made up of 16 cylinders positioned inside a cube of size _LxLxL_, as illustrated in the picture below:

[image]

I initially created the cylinders using Part::Feature objects as follows:
```
def addCylinderUsingMakeCylinder(start_point=[0,0,0], direction=[1,1,1], length=1, radius=0.1, name='Name'):
	'''
	Add cylinder based on start point, direction and length, using FreeCAD.ActiveDocument.addObject("Part::Feature", name).
	'''
	x = direction[0]
	y = direction[1]
	z = direction[2]
	u = Base.Vector(x,y,z).normalize()
	cyl_shape = Part.makeCylinder(radius, length, Base.Vector(start_point[0],start_point[1], start_point[2]), u, 360)
	cyl_obj = FreeCAD.activeDocument().addObject("Part::Feature", name)
	cyl_obj.Shape = cyl_shape
	return(cyl_obj)
```
All cylinders are then fused together using a Part::MultiFuse object.

However, when increasing the size of the unit-cell cube L from 1 to 100, some cylinders are ignored. See figure below for L=100:
[image]

### Workaround
I managed to fix this issue using Part::Cylinder objects instead, using a function like this:
```
def addCylinderBasedOnDirection(start_point=[0,0,0], direction=[1,1,1], length=1, radius=0.1, name='Name'):
	'''
	Add cylinder based on start point, direction and length, using FreeCAD.ActiveDocument.addObject("Part::Cylinder", name).
	'''
	obj = FreeCAD.ActiveDocument.addObject("Part::Cylinder", name)
	z = np.array([0,0,1])
	c = np.dot(z,direction)/(np.linalg.norm(z)*np.linalg.norm(direction))
	angle_rad = np.arccos(c)
	angle_deg = np.rad2deg(angle_rad)
	rotation_axis = np.cross(z, direction)
	obj.Placement = FreeCAD.Placement(FreeCAD.Vector(*start_point), FreeCAD.Rotation(FreeCAD.Vector(*rotation_axis), angle_deg))
	obj.Height = length
	obj.Radius = radius
	return obj
```

### FreeCAD file showing the problem+workaround and full macro used to generate it
[boolean-union-failure.zip](https://github.com/user-attachments/files/17577530/boolean-union-failure.zip)

Full macro text for convenience:
```
import FreeCAD
from FreeCAD import Base
import Part
import numpy as np

def addCylinderUsingMakeCylinder(start_point=[0,0,0], direction=[1,1,1], length=1, radius=0.1, name='Name'):
	'''
	Add cylinder based on start point, direction and length, using FreeCAD.ActiveDocument.addObject("Part::Feature", name).
	'''
	x = direction[0]
	y = direction[1]
	z = direction[2]
	u = Base.Vector(x,y,z).normalize()
	cyl_shape = Part.makeCylinder(radius, length, Base.Vector(start_point[0],start_point[1], start_point[2]), u, 360)
	cyl_obj = FreeCAD.activeDocument().addObject("Part::Feature", name)
	cyl_obj.Shape = cyl_shape
	return(cyl_obj)

def addCylinderFromAtoB(A=[0,0,0], B=[1,1,1], radius=0.1, name='Name'):
	'''
	Add cylinder based on start and end point, using FreeCAD.ActiveDocument.addObject("Part::Cylinder", name).
	'''
	obj = FreeCAD.ActiveDocument.addObject("Part::Cylinder", name)
	direction = np.array(B)-np.array(A)
	z = np.array([0,0,1])
	c = np.dot(z,direction)/(np.linalg.norm(z)*np.linalg.norm(direction))
	angle_rad = np.arccos(c)
	angle_deg = np.rad2deg(angle_rad)
	rotation_axis = np.cross(z, direction)
	location = FreeCAD.Vector(*A)
	obj.Placement = FreeCAD.Placement(location, FreeCAD.Rotation(FreeCAD.Vector(*rotation_axis), angle_deg))
	obj.Height = np.linalg.norm(direction)
	obj.Radius = radius
	return obj

def addCylinderBasedOnDirection(start_point=[0,0,0], direction=[1,1,1], length=1, radius=0.1, name='Name'):
	'''
	Add cylinder based on start point, direction and length, using FreeCAD.ActiveDocument.addObject("Part::Cylinder", name).
	'''
	obj = FreeCAD.ActiveDocument.addObject("Part::Cylinder", name)
	z = np.array([0,0,1])
	c = np.dot(z,direction)/(np.linalg.norm(z)*np.linalg.norm(direction))
	angle_rad = np.arccos(c)
	angle_deg = np.rad2deg(angle_rad)
	rotation_axis = np.cross(z, direction)
	obj.Placement = FreeCAD.Placement(FreeCAD.Vector(*start_point), FreeCAD.Rotation(FreeCAD.Vector(*rotation_axis), angle_deg))
	obj.Height = length
	obj.Radius = radius
	return obj

def tetra(loc=[0,0,0], a=[1,1,1], rod_radius=0.1, method=2):
	'''
	Create a single tetrahedral arrangement of cylinders.
	'''
	ax, ay, az = a
	rod_length = np.linalg.norm(np.array(a)/4)
	if method == 0:
		print('addCylinderUsingMakeCylinder')
		cyl0 = addCylinderUsingMakeCylinder(start_point=loc, direction=[ ax, ay,-az], length=rod_length, radius=rod_radius, name=f'method-{method}-ppm')
		cyl1 = addCylinderUsingMakeCylinder(start_point=loc, direction=[-ax,-ay,-az], length=rod_length, radius=rod_radius, name=f'method-{method}-mmm')
		cyl2 = addCylinderUsingMakeCylinder(start_point=loc, direction=[ ax,-ay, az], length=rod_length, radius=rod_radius, name=f'method-{method}-pmp')
		cyl3 = addCylinderUsingMakeCylinder(start_point=loc, direction=[-ax, ay, az], length=rod_length, radius=rod_radius, name=f'method-{method}-mpp')
	elif method == 1:
		print('addCylinderFromAtoB')
		cyl0 = addCylinderFromAtoB(A=loc, B=np.array(loc)+np.array([ ax, ay,-az])/4, radius=rod_radius, name=f'method-{method}-ppm')
		cyl1 = addCylinderFromAtoB(A=loc, B=np.array(loc)+np.array([-ax,-ay,-az])/4, radius=rod_radius, name=f'method-{method}-mmm')
		cyl2 = addCylinderFromAtoB(A=loc, B=np.array(loc)+np.array([ ax,-ay, az])/4, radius=rod_radius, name=f'method-{method}-pmp')
		cyl3 = addCylinderFromAtoB(A=loc, B=np.array(loc)+np.array([-ax, ay, az])/4, radius=rod_radius, name=f'method-{method}-mpp')
	else:
		print('addCylinderBasedOnDirection')
		cyl0 = addCylinderBasedOnDirection(start_point=loc, direction=[ ax, ay,-az], length=rod_length, radius=rod_radius, name=f'method-{method}-ppm')
		cyl1 = addCylinderBasedOnDirection(start_point=loc, direction=[-ax,-ay,-az], length=rod_length, radius=rod_radius, name=f'method-{method}-mmm')
		cyl2 = addCylinderBasedOnDirection(start_point=loc, direction=[ ax,-ay, az], length=rod_length, radius=rod_radius, name=f'method-{method}-pmp')
		cyl3 = addCylinderBasedOnDirection(start_point=loc, direction=[-ax, ay, az], length=rod_length, radius=rod_radius, name=f'method-{method}-mpp')
	return [cyl0, cyl1, cyl2, cyl3]

def createRCD(a=[1,1,1], name='RCD', method=2):
	'''
	Create a Rod-Connected Diamond unit-cell.
	'''
	
	ax, ay, az = a
	rn = 0.1
	r = rn*az
	
	t0 = tetra(loc=np.multiply([3/4, 1/4, 1/4],a), a=a, rod_radius=r, method=method)
	t1 = tetra(loc=np.multiply([1/4, 3/4, 1/4],a), a=a, rod_radius=r, method=method)
	t2 = tetra(loc=np.multiply([1/4, 1/4, 3/4],a), a=a, rod_radius=r, method=method)
	t3 = tetra(loc=np.multiply([3/4, 3/4, 3/4],a), a=a, rod_radius=r, method=method)

	obj_list = t0 + t1 + t2 + t3

	f = FreeCAD.activeDocument().addObject("Part::MultiFuse","Fusion")
	f.Shapes = obj_list
	f.Label = name

def main():
	# create new document
	doc = FreeCAD.newDocument()

	# select existing document
	# doc = FreeCAD.activeDocument()
	
	# remove all current objects in document
	#for i in doc.Objects:
	#  doc.removeObject(i.Name)

	# create example geometries
	for L in [1,10,100]:
		for method in [0,1,2]:
			createRCD(a=[L,L,L], name=f'RCD_{L}_{method}', method=method)

	doc.recompute()

if __name__ == "__main__":
	main()
```

### Possibly relevant FreeCAD settings
- Unit system: mm, kg, s, degree
- Number of decimals: 12

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * boltsfc 2022.11.5
  * CrystalWorkbench
  * OpticsWorkbench 1.0.8
  * PythonWorkbenchTemplate
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-30:
@mtav can you edit the OP and add your full [About](https://wiki.freecad.org/About) info to the end of it ?

maxwxyz, 2024-10-30:
@mtav can you try this in the latest weekly as booleans had a lot of changes lately: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
TIA

mtav, 2024-10-31:
Yes, it works with the latest weekly build! Thank you!

When I first started it, I got a lot of errors, so I simply renamed my %APPDATA%\FreeCAD directory and restarted FreeCAD.

It asked me to recompute the document when I opened it, which instantly fixed the missing cylinders.

For the script, I was missing numpy at first, so I made some changes for the script to work without it and then it worked fine. No missing cylinders.

Version used:

> OS: Windows 11 build 22631
> Word size of FreeCAD: 64-bit
> Version: 1.1.0dev.39057 (Git)
> Build type: Release
> Branch: main
> Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
> Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
> Locale: English/United Kingdom (en_GB)
> Stylesheet/Theme/QtStyle: unset/unset/Qt default
> 

Here is the numpy-free script, with some improvements (I also tested intersections with boxes, which worked correctly):
```
import FreeCAD
from FreeCAD import Base
import Part
import math

def addCylinderUsingMakeCylinder(start_point=[0,0,0], direction=[1,1,1], length=1, radius=0.1, name='Name'):
	'''
	Add cylinder based on start point, direction and length, using FreeCAD.ActiveDocument.addObject("Part::Feature", name).
	'''
	x = direction[0]
	y = direction[1]
	z = direction[2]
	u = Base.Vector(x,y,z).normalize()
	cyl_shape = Part.makeCylinder(radius, length, Base.Vector(start_point[0],start_point[1], start_point[2]), u, 360)
	cyl_obj = doc.addObject("Part::Feature", name)
	cyl_obj.Shape = cyl_shape
	return(cyl_obj)

def addCylinderFromAtoB(A=[0,0,0], B=[1,1,1], radius=0.1, name='Name'):
	'''
	Add cylinder based on start and end point, using FreeCAD.ActiveDocument.addObject("Part::Cylinder", name).
	'''
	obj = FreeCAD.ActiveDocument.addObject("Part::Cylinder", name)
	direction = Base.Vector(B)-Base.Vector(A)
	z = Base.Vector([0,0,1])
	angle_deg = math.degrees(z.getAngle(direction))
	rotation_axis = z.cross(direction)
	location = FreeCAD.Vector(*A)
	obj.Placement = FreeCAD.Placement(location, FreeCAD.Rotation(FreeCAD.Vector(*rotation_axis), angle_deg))
	obj.Height = direction.Length
	obj.Radius = radius
	return obj

def addCylinderBasedOnDirection(start_point=[0,0,0], direction=[1,1,1], length=1, radius=0.1, name='Name'):
	'''
	Add cylinder based on start point, direction and length, using FreeCAD.ActiveDocument.addObject("Part::Cylinder", name).
	'''
	obj = FreeCAD.ActiveDocument.addObject("Part::Cylinder", name)
	direction = Base.Vector(direction)
	z = Base.Vector([0,0,1])
	angle_deg = math.degrees(z.getAngle(direction))
	rotation_axis = z.cross(direction)
	obj.Placement = FreeCAD.Placement(FreeCAD.Vector(*start_point), FreeCAD.Rotation(FreeCAD.Vector(*rotation_axis), angle_deg))
	obj.Height = length
	obj.Radius = radius
	return obj

[... 61 lines cut ...]

if __name__ == "__main__":
	createNewDoc = True
	if createNewDoc:
		# create new document
		doc = FreeCAD.newDocument()
	else:
		# select existing document
		doc = FreeCAD.activeDocument()
	
		# remove all current objects in document
		for i in doc.Objects:
			doc.removeObject(i.Name)

	# create example geometries
	for L in [1,10,100,1000]:
		for method in [0,1,2]:
			createRCD(a=[L,L,L], name=f'RCD_{L}_{method}', method=method)

	doc.recompute()
```

I also learned today about [Part.makeCompound](https://forum.freecad.org/viewtopic.php?t=45467), which might possibly be an even better way to do it?

Looking forward to the next official FreeCAD release!

closed 2024-10-31 by mtav