FreeCAD/FreeCAD #17585: No way to globally change line width and point size

Beep6581, 2024-10-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There seems to be no way of globally setting the line width and point size of a sketch in Preferences.

I can select a sketch and change the line width and point size in the "View" tab, but that applies only to the selected sketch.

With the default point size of 2, it's impossible to see whether I'm hovering over a point on a high resolution screen.

[image]

[image]

[image]

### Full version info

```shell
`b6cda4309a9031bebffa73b694e9a9cc125d64de`
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-30:
Thanks for the ticket, @Beep6581. However we really do need the full [About](https://wiki.freecad.org/About) info due to x-reffing what 3rd party dependencies the build was built against. TIA

maxwxyz, 2024-10-31:
Line width can be changed in the preferences: Sketcher>Display>View scale ratio.
There are some setting in the part design preferences for line width and vertex size and I think a setting in draft/bim for choosing document settings.

maxwxyz, 2024-10-31:
@kadet1090 FYI

kadet1090, 2024-10-31:
Can do, I think that I have good and universal solution for problems like this in mind.

hyarion, 2024-11-01:
Note that only a line width of 1 is guaranteed by OpenGL.

https://registry.khronos.org/OpenGL-Refpages/gl4/html/glLineWidth.xhtml

> Not all widths can be supported when line antialiasing is enabled. If an unsupported width is requested, the nearest supported width is used. Only width 1 is guaranteed to be supported; others depend on the implementation. Likewise, there is a range for aliased line widths as well.

Beep6581, 2024-11-01:
@luzpaz no worries, it was probably a build a few commits older than the one I'm running now:

OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39046 (Git)
Build type: Release
Branch: main
Hash: b6cda4309a9031bebffa73b694e9a9cc125d64de
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * OpenDark 2023.12.17
  * fasteners 0.5.25
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.9
  * Curves 0.6.42

@maxwxyz I see. "View scale ratio" seems to only apply to lines, as points get obliterated:

[image]

[image]

maxwxyz, 2024-11-08:
FYI the point size can be changed by the marker size in the preferences
See this demo:
https://youtu.be/BJG1SaANUa0?si=UlGsVzCiLrzyP1Nd&t=376

Better sorted preferences would be great.

Beep6581, 2024-11-11:
@maxwxyz that setting does not apply to all "markers", specifically it does not apply to external geometry.

[image]

[image]