FreeCAD/FreeCAD #17587: Legend behind results when displaying SolverElmerResult

oladec, 2024-10-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I ran the FEM example "Electrostatics Capacitance Two Balls". 
When I get the "SolverElmerResult" it covers the legend. This is annoying since it is hard to see the numbers corresponding to the color when zooming in to the details in the example.

### Full version info

```shell
Tested with two versions and safe mode which behaves the same:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * A2plus.backup1695819458.3926806 0.4.60d (Disabled)
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * Cfd
  * CfdOF 1.27.11
  * EM 2.1.1
  * ExplodedAssembly
  * Help 1.0.3
  * kicadStepUpMod 11.3.9
  * OpenDark 2024.3.13
  * pcb 6.2023.1
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * A2plus.backup1695819458.3926806 0.4.60d (Disabled)
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * Cfd
  * CfdOF 1.27.11
  * EM 2.1.1
  * ExplodedAssembly
  * Help 1.0.3
  * kicadStepUpMod 11.3.9
  * OpenDark 2024.3.13
  * pcb 6.2023.1
```

### Subproject(s) affected?

FEM

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-31:
Related: https://github.com/FreeCAD/FreeCAD/issues/11665
@FEA-eng FYI

FEA-eng, 2024-10-31:
Yeah, I know about that. The legend always stays behind, regardless of the solver:

[image]

It's yet another issue with that color bar - the main problem is that it might be covered by the navigation cube or jump to the right so much that all labels disappear (this usually happens when saving). Unfortunately, this color bar is really problematic to fix. Both Uwe and Werner already worked on it in the past and it's still not working properly: https://github.com/FreeCAD/FreeCAD/issues/6047#issuecomment-1445387153

@marioalexis84 FYI

FEA-eng, 2026-11-02:
Still relevant, also on Windows.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```