FreeCAD/FreeCAD #17588: topologischen Benennung scheint nicht zu funktionieren / Topological naming problem does not seem to work

HKaufmann2, 2024-10-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Vorhandesnes Teil geöffnet. Fehler "Neuberechnung erforderlich". Diese Durchgeführt. 
Am Anfang von einem Parameter der Aufpolsterung Länge geändert.
Danach Fehler bei Champfer. 
Wenn die Neuberechnung nicht durchgeführt wird klappt die Änderung der Länge.

Existing part opened. Error "Recalculation required". This performed. 
At the beginning of a parameter of the padding length changed.
After that, Champfer error. 
If the recalculation is not performed, the length change will be done.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-31:
Can you try this with the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Can you share the example file please.

HKaufmann2, 2024-10-31:
[Ivar_RB_Zus.zip](https://github.com/user-attachments/files/17585591/Ivar_RB_Zus.zip)
Hoffe das die Datei rüberkommt. Hope the file comes across.

maxwxyz, 2024-10-31:
I cannot test as there are unsupported features and errors when opening the file. With which version did you create the file and what assembly addon did you use (as your about info does not show any addons).

HKaufmann2, 2024-10-31:
[Ivar_RB_Stuetzbeine.zip](https://github.com/user-attachments/files/17587383/Ivar_RB_Stuetzbeine.zip)
Habe die Datei mit FreeCAD 0.21.2 + Part Design erstellt.
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * ArchTextures
  * Assembly4 0.50.15
  * BillOfMaterials 1.0.1
  * BIM021 2021.12.0
  * boltsfc 2022.11.5
  * btl 0.9.9
  * CADExchanger
  * CommandPanel
  * CurvedShapes 1.0.12
  * Curves 0.6.51
  * Defeaturing 1.2.2
  * ExplodedAssembly
  * fasteners 0.5.29
  * Fonts
  * freecad.gears 1.3.0
  * Help 1.0.3
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * offline-documentation 1.0.0-alpha
  * parts_library
  * pcb 6.2023.1
  * sheetmetal 0.5.3
  * symbols_library
  * TitleBlock 0.5.2.2
  * toSketch 1.0.1
  * Trails 2022.1.0
  * WB_Organizer 2024.1.29
  * woodworking 0.21.2.33771
  * workfeature
[/code]

ElPatoNoEsTu, 2024-10-31:
Topological Naming is not adressed in  FreeCAD 0.21.2  please try using actual weekly or V1.0RC2  -  In der von Dir verwendeten Version ist die TNP Behandlung noch nicht drinne.

maxwxyz, 2024-10-31:
issue was reported with the 1.0 RC

maxwxyz, 2025-07-27:
is this still an issue in the latest dev version https://github.com/FreeCAD/FreeCAD/releases/weekly-2025.07.26/

closed 2025-08-24 by github-actions[bot]