FreeCAD/FreeCAD #17589: Feature Request   Manually making everything recompute is also confusing to do, would be nice if we had a "recompute everything" command, this off topic here though. Could someone create an issue about making recompute easier and having guidelines for testing models? Ouw comments here should probably be marked as off topic

AIRCAP, 2024-10-31:
There should be a key combo or menu entry shortcut for the action of
1. right clicking on the active document "mark all for recompute"
2. thendoing App.ActiveDocument.recompute()

yes its trivial to add that as a user macro button, but why not have shift+ctrl+r do that per default?

>              Manually making everything recompute is also confusing to do, would be nice if we had a "recompute everything" command, this off topic here though. Could someone create an issue about making recompute easier and having guidelines for testing models? Ouw comments here should probably be marked as off topic

_Originally posted by @adrianinsaval in https://github.com/FreeCAD/FreeCAD/issues/17543#issuecomment-2449521991_

luzpaz, 2024-10-31:
The key combo being `shift+ctrl+r` makes total sense to me since that is the key combo for hard refreshing browsers and would be an intuitive carry-over into FreeCAD

AIRCAP, 2024-10-31:
@luzpaz wanna add "Good first issue?" to encourage dev newbees? I think that should be a relatively easy one to implement and get into it.

kaiwas, 2024-10-31:
Сan also modify the macro to recalculate the selected objects.

But I agree that these actions "out of the box" within the program itself are very much lacking.

AIRCAP, 2024-10-31:
i think you can assign macros to buttons on custom toolbars, and also assign keyboard shortcuts to them, but this doesn't change that this kinda should come with the base program, especially since its workbench independent functionality

chesterrc, 2024-11-19:
Hi could I work on this?

adrianinsaval, 2024-11-19:
> Hi could I work on this?

of course, contributors are always welcome!

chesterrc, 2024-11-19:
> > Hi could I work on this?
> 
> of course, contributors are always welcome!

would you be able to point me to a file to start looking into?  Or the general direction on how I could start?

maxwxyz, 2026-11-07:
This is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/658211dd7060fff18d08a4bb6b7b14fa5998d6c6

Recompute All is in the Edit menu with Shift+F5. It marks every object in the active document and recomputes it.