FreeCAD/FreeCAD #17590: FemMesh Netgen error

flamatt1966, 2024-10-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While trying to mesh an object in got To use FemMesh Netgen objects, install the Netgen Python bindingsTraceback (most recent call last): File "", line 4, in File "/Applications/FreeCAD.app/Contents/Resources/Mod/Fem/femmesh/netgentools.py", line 122, in run_netgen import pyngcore as ngcore : No module named 'pyngcore' 
nd mwsh hlted

### Full version info

```shell
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2024.7.24
  * freecad.gears 1.0.0
  * OpenTheme 2024.9.1
  * Dracula 0.0.5
  * Render 2023.8.18
  * Manipulator 1.5.0
  * fasteners 0.5.29
  * Lithophane
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.2
  * ExtremeProLight 1.6.0
  * Curves 0.6.15
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-31:
@flamatt1966 please test FEM related issues on weekly (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) and report back. Reason: lots of FEM fixes won't make it into 1.0

luzpaz, 2024-10-31:
Related to #16759

marioalexis84, 2024-10-31:
This issue has been fixed

closed 2024-10-31 by maxwxyz

marioalexis84, 2024-10-31:
https://github.com/FreeCAD/FreeCAD/pull/17011

luzpaz, 2024-10-31:
@flamatt1966 you'll either need to use the weekly or wait for release of 1.0 (unless there is an RC3)

marioalexis84, 2024-10-31:
You can also use the new meshing implementation for Netgen by installing the netgen-mesher package from pip using the FreeCAD Python interpreter. If your computer is not too old, that package will work.