FreeCAD/FreeCAD #17591: FEM: Freeze at 100% CPU when searching mesh for solid (without meshgroups)

AIRCAP, 2024-10-31:
This bug is split of #17427 to address the 100% CPU freeze bug independent from the meshgroup regression.

> It's actually two bugs for the price of one: 
> 1. Searching for solid ID's in certain meshes takes forever - at 100% CPU load ... I don't know if it takes literally forever, but I have waited over an hour once because I forgot about it and it still hadn't progressed - this only happens when meshgroups are disabled. This bug is old, it was already there in 0.21.2
> 2. The regression: ... addressed by #17427
_Originally posted by @AIRCAP in https://github.com/FreeCAD/FreeCAD/issues/17427#issuecomment-2436197131_

for details and files to reproduce see #17427

AIRCAP, 2024-10-31:
Crosspost:

I've been looking into the 100% CPU load side of things a bit more (Without Mesh Regions)

Here is a much simpler file

[smaller.zip](https://github.com/user-attachments/files/17579822/smaller.zip)

it now can write the inp file:

> Get mesh data for constraints, materials and element geometry...
> Materials
> Constraint: MaterialSolid --> We're going to search in the mesh for the element ID's.
> Constraint: MaterialSolid001 --> We're going to search in the mesh for the element ID's.
>     ReferenceShape ... Type: Solid, Object name: BooleanFragments, Object label: BooleanFragments, Element name: Solid1
> **<here is where it takes forever>**    ReferenceShape ... Type: Solid, Object name: BooleanFragments, Object label: BooleanFragments, Element name: Solid2
> **<here is where it takes forever again>**   binary search: get_femelements_by_femnodes_bin
> len femnodes_ele_table: 19167
> found Volumes: 4096
> Count finite elements as sum of constraints:   9922
> Count finite elements of the finite element mesh: 9922
> ConstraintFixed:
>     Type: Fem::ConstraintFixed, Name: ConstraintFixed
>     ReferenceShape ... Type: Face, Object name: BooleanFragments, Object label: BooleanFragments, Element name: Face161
>     ReferenceShape ... Type: Face, Object name: BooleanFragments, Object label: BooleanFragments, Element name: Face88
> ConstraintForce:
>     Type: Fem::ConstraintForce, Name: ConstraintForce
>     ReferenceShape ... Type: Face, Object name: BooleanFragments, Object label: BooleanFragments, Element name: Face490
> Error on: ConstraintForce --> BooleanFragments.Face490
>   sum_node_areas: 1538.6418339695256
>   ref_face_area:  1557.913072171677
> Deviation sum_node_load to frc_obj.Force is more than 1% :  0.9876300940364476
>   sum_ref_face_node_area: 1538.6418339695256
>   sum_ref_face_area:      1557.913072171677
>   sum_node_load:          98.76300940364476
>   frc_obj.Force:          100.0
>   the reason could be simply a circle area --> see method get_ref_face_node_areas
>   the reason could also be a problem in retrieving the ref_face_node_area
> **Getting mesh data time: 10551.376 seconds.**
> 
> CalculiX solver input writing...
> Input file:/tmp/fcfem_w86r2eev/FEMMeshGmsh.inp
> One monster input file.
> Writing time CalculiX input file: 0.185 seconds.
> 

but it is still excessively long, you probably don't want to wait that long ;)
also of course with the mesh so low res the Jacobian is kinda negative. If I make the mesh fine it works (as long as I don't assign any materials to solids, because if I do it takes even foreverer)

in this example there are no BOP errors in the file that leads to the mesh - should pass all geometry checks.

_Originally posted by @AIRCAP in https://github.com/FreeCAD/FreeCAD/issues/17427#issuecomment-2448601778_

AIRCAP, 2024-10-31:
for comparison, with meshgroups the meshing takes marginally longer, but thats just a few seconds, while the search for the solids can take on the order of several hours (depending on how many cpu cores you have, it is parallelized)

with faces, edges and vertices this isn't the case, they are found quickly.

FEA-eng, 2024-10-31:
@marioalexis84 FYI

marioalexis84, 2024-11-19:
@maxwxyz this is not a bug, but rather a request for improved algorithm performance.

luzpaz, 2026-09-29:
Renamed ticket  to more accurately reflect request