FreeCAD/FreeCAD #17593: FEM: Gui: ColorLegend scale/options completely broken

AIRCAP, 2024-10-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In FEM, when visualizing the results with color coding, the Legend does not accept changes in its display range under some conditions and instead reverts to stupid values (-0.5 to +0.5)

possibly related to #6950

Example:
[BrokenColorLegend.zip](https://github.com/user-attachments/files/17589340/BrokenColorLegend.zip)

[image]

1st issue: right clicking on the color legend and clicking on **options** :  no effect

2nd issue: double clicking on the color legend opens the legend options window. If changing any value and then clicking elsewhere - or OK changes the scale of the color legend from -0.5 to +0.5 - regardless which values were entered - this also happens when clicking anywhere outside of the options window after changing any value - even outside of the FreeCAD window

[image]

3rd issue: Clicking on the part (in this case the "Cut") in either tree or 3d view resets the color legend to defaults (in this case 0.000 to 1.4e+7)

### Full version info

```shell
Current main:

OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39099 (Git)
Build type: Debug
Branch: main
Hash: b7306d29e35aab9ebc0c9ef61e9b9d26c6537bcd
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.12
  * CfdOF 1.27.13
  * Curves 0.6.51
  * AirPlaneDesign 0.4.1
```

### Subproject(s) affected?

FEM

### Anything else?

For debugging: The reproduction of this issue is related to the tree of elements in the visualization pipeline.

[image]

Even if they are not visible, their Gui Properties in Coloring (Field and Vector Mode) have an effect on the behaviour. In this case, the jump to -0.5 to +0.5 comes from that being the values of the mother item in the tree (the pipeline itself) which has the field set to none and so does the WarpVector.  if one of these items is set to another Field - for example directional strain or displacement, the min and max value of that field affect what the legend "jumps" to in the currently visible part.

a half-way workaround is to set all parts to display the same field. (in this case that would be "von Mises Stress") - however that does not fix the 3rd issue that whatever the legend is set to resets if a part is selected accidently or on purpose (single left mouse button in 3d view is enough)

as is this makes the color visualization pretty useless.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-10-31:
pinging @FEA-eng because FEM ;)

FEA-eng, 2024-10-31:
> Is it the same as #15013 Is color legend working for the pipeline objects or if you double click on the CCX_results object? IIRC the legend is only for one of them.

Back then it wasn't working at all, now it only seems to reset under specific circumstances. There's a known issue that the manual ranges are getting reset when switching to a different object.

Yes, the legend is shown only for the pipeline, the CCX_Results object has no legend, only a color map.

@marioalexis84 FYI

AIRCAP, 2024-10-31:
to me it looks like the pipeline objects (including the pipeline itself) are fighting over the legend, and the one that is currently visible and active/selected isn't always the one winning (thus the reset to +-0.5)

an issue is that this doesn't just affect the legend itself but also the coloration of the pipeline object, which makes this extra annoying. (the color range has to be adjusted to inspect regions with lower strain, especially in multi-material parts because the weaker materials also have weaker yield-strength)

it would be kinda useful if the legend had a logarithmic scale setting for the coloration, especially for strain

maxwxyz, 2025-03-18:
This is still an issue. Just opening the color gradient scale options and pressing cancel lets my model appear completely red (I did not change any parameter). Clicking on the warp object resets the coloring.

https://github.com/user-attachments/assets/145e555d-3349-435e-b392-759fd168b6cd

Also, toggling the visibility of the parent pipeline object, changes the values of the color gradient without changing color on the model:

https://github.com/user-attachments/assets/9eb103f2-e557-4c69-bce7-fb68d6629915