FreeCAD/FreeCAD #17596: Cannot fillet a padded cylinder - "Fillet not possible on selected shapes"

ernstschnell75, 2024-10-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to put a 2 mm fillet on the outside top edge of the cylinder in the back (padded from two concentric circles) results in a "Fillet not possible on selected shape" error, which seems strange given that it is a simple circle. No integrity issues are obvious on the model.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * A2plus 0.4.68
  * sheetmetal 0.5.3
```

### Subproject(s) affected?

PartDesign

### Anything else?

There are a number of issues with fillets and chamfers listed, however I did not find any resulting in this particular error.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-31:
I cannot reproduce this. Can you share a file where it happens?

depthoffocus, 2024-10-31:
Just a passing thought:

The version being reported here is 1.0RC2, which is before #16920 was fixed?

I wonder if the issue here is that the edge selection on screen did not match the internal selection, because of this bug?

@ernstschnell75 I wonder if you are able to reproduce this on the latest weekly build?

ernstschnell75, 2024-10-31:
I will try once I get home.

ernstschnell75, 2024-11-01:
So first things first, here is the model attached. Sorry, I had it attached to the initial report, but it seems that github only allows one attachment at a time. Noob mistake :-(
[Exercise 43.zip](https://github.com/user-attachments/files/17604061/Exercise.43.zip)

ernstschnell75, 2024-11-01:
I have run the same model with the latest Windows weekly build. It now comes back with a general error message and a python message in the console that this shape. My issue is that it should be, this still just being a padded double circle, and hence a regular circular edge.
[image]

ernstschnell75, 2024-11-01:
I probably should have mentioned, the weekly build I tried was
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * A2plus 0.4.68
  * sheetmetal 0.5.3

depthoffocus, 2024-11-01:
OK so it's not the problem I was wondering about.

You do have a broken fillet in your design -- Fillet001. (I don't know why)

[image]

If you delete that fillet, you can put a 2mm fillet on that edge just fine:

[image]

depthoffocus, 2024-11-01:
The edge in the broken fillet, Edge75, turns out to be the seam of that cylinder. That is weird. 

I wonder if this is a TNP issue -- that is, it was a different edge you chamfered before you made some change to the underlying sketches, and it didn't get renamed properly.

closed 2025-09-09 by github-actions[bot]