FreeCAD/FreeCAD #17600: Arch: GUI element imported too early in `Arch.makeWindow` which errors when running headless

luzpaz, 2024-11-01:
### Discussed in https://github.com/FreeCAD/FreeCAD/discussions/17598

<div type='discussions-op-text'>

<sup>Originally posted by **rix133** October 31, 2024</sup>
The GUI component is imported before needed [from DraftGui import todo](https://github.com/FreeCAD/FreeCAD/blame/8865450a3e14220925e0e449c0f1f79056b4fb89/src/Mod/BIM/Arch.py#L1089)   so if run without GUI one will get an error 

A solution would be to add this import after the GUI check eg after [if obj.Base and FreeCAD.GuiUp:](https://github.com/FreeCAD/FreeCAD/blame/8865450a3e14220925e0e449c0f1f79056b4fb89/src/Mod/BIM/Arch.py#L1137)</div>

luzpaz, 2024-11-01:
@rix133 feel free to submit a PR

luzpaz, 2024-11-01:
Can you also provide a short example script that trigger warning?

Roy-043, 2024-11-01:
I am not sure about the proposed solution. For one thing the todo function is not defined in the DraftGui module. Secondly, the same import statement occurs in other places in the BIM code. I suggest looking at all of them.

rix133, 2024-11-01:
> Can you also provide a short example script that trigger warning?

under windows I can reproduce it like this (using the RC2 build for windows):
`C:\FreeCAD_1.0.0RC2-conda-Windows-x86_64-py311> .\bin\python.exe -m unittest .\Mod\BIM\TestArch.py`

i.e all the test won't run headless

### Test Output Summary

**Errors:**

1. **testBuildingPart**
   - **Description:** Create a BuildingPart from a wall with a window and check its shape.
   - **Error:** `AttributeError: module 'FreeCADGui' has no attribute 'UiLoader'`
   - **Traceback:**
     ```plaintext
     File "TestArch.py", line 712, in testBuildingPart
       win = Arch.makeWindow(sk)
     File "Arch.py", line 1089, in makeWindow
       from DraftGui import todo
     ```

2. **testRebar**
   - **Error:** `ImportError: Cannot load Gui module in console application.`
   - **Traceback:**
     ```plaintext
     File "TestArch.py", line 456, in testRebar
       r = Arch.makeRebar(s, sk, diameter=.1, amount=2)
     File "Arch.py", line 545, in makeRebar
       import ArchRebar
     ```

3. **testWindow**
   - **Error:** `AttributeError: module 'FreeCADGui' has no attribute 'UiLoader'`
   - **Traceback:**
     ```plaintext
     File "TestArch.py", line 496, in testWindow
       win = Arch.makeWindow(sk)
     File "Arch.py", line 1089, in makeWindow
       from DraftGui import todo
     ```

**Summary:**
- **Ran:** 24 tests in 10.249s
- **Failed:** 3 errors

Disclamer: I'm totally new to this codebase

yorikvanhavre, 2024-11-05:
> C:\FreeCAD_1.0.0RC2-conda-Windows-x86_64-py311> .\bin\python.exe -m unittest .\Mod\BIM\TestArch.py

Interesting, I never thought of running these tests like that... They were mainly made for GUI usage and are not made to run without it, but it shouldn't be hard to adapt. Maybe for some of them it makes little sense to run without the GUI too.  I'll look into it