FreeCAD/FreeCAD #17602: Some icons appear low resolution on 4K screen

frysztak, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using a 4K screen with 2x scaling, some icons (chevrons, recompute, formula) appear to be blurry:

[image]
[image]

### Full version info

```shell
OS: Arch Linux (GNOME/gnome/wayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.39020 (Git)
Build type: Release
Branch: makepkg
Hash: 52982c0c67d09cde4da5eff267c9c0cc063d456a
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Curves 0.6.47
  * ThreadProfile 1.92.0
  * fasteners 0.5.29
  * A2plus 0.4.67
  * Pyramids-and-Polyhedrons
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-01:
Maybe related:
- https://github.com/FreeCAD/FreeCAD/issues/13118
- https://github.com/FreeCAD/FreeCAD/issues/12992
- https://github.com/FreeCAD/FreeCAD/issues/6765

benj5378, 2025-03-13:
@captain0xff currently working on this

closed 2025-04-04 by kadet1090