FreeCAD/FreeCAD #17603: Error when typing equation into properties of an object

semhustej, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- Open new file
- Create Draft_Circle
- Type '10/2' to the Radius property of the circle and hit 'Enter' 
- FreeCAD shows error after the '/' is typed
```
09:58:28  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ArithmeticError'>: Not matching Unit!
```

- the value from the equation is calculated properly

See behavior in video:

https://github.com/user-attachments/assets/43716f23-1811-4006-a313-16f250aebff3

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Draft

### Anything else?

I am able to reproduce the issue in FC 0.21.2

I am able to reproduce the issue with other types of objects and different properties (Draft_Rectangle, Arch_Wall)

The error does not limit the functionality of FreeCAD.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-01:
@Roy-043 FYI

Roy-043, 2024-11-01:
This has nothing to do with Draft. It is a Core issue. Entering `5/2 mm` is interpreted as `5/(2 mm)`. Hence the error message. We see the same issue in the handling of expressions.

leoheck, 2026-10-29:
Related: cannot divide directly in the field anymore.
It seems the equations are a very poorly maintained feature. It is full of little bugs.
It used to work better.
https://github.com/user-attachments/assets/aa06a771-6848-4d01-8f1c-3a92c7229621

JohnOCFII, 2026-10-29:
Likely related to this active blocker: https://github.com/FreeCAD/FreeCAD/issues/33388