FreeCAD/FreeCAD #17604: Chamfer: BRep_API: command not done - on a bspline wall

it-guru, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have reduce the problem on a simple geometry - i call it bspline wall (thinkness of the wall is 2mm).
[BSplineChamferBug.FCStd.zip](https://github.com/user-attachments/files/17597840/BSplineChamferBug.FCStd.zip)

I need a chamfer on the upper __inner__ edge.

If i create the chamfer on the __outer__ edge, all works fine:
[image]

If i try to create the chamfer on the __inner__ edge ...
[image]
... it fails with the error output:
**Chamfer: BRep_API: command not done**

The problem seems not to be related to the relation between "wall thinkness" and the chamfer length. I have try the chamfer lenght of 0,1mm - 1,0mm and all produces the same error.
The problem seems to be related to the base geometry of the "wall" (extrution), because in some cases the inner edge is posible and the outer edge chamfer fails - in some cases the inner edge chamfer is posible/working.

In the Freecad 0.21.2 it seems to work on the bspline-eight base geometry:
[BSplineChamfer0_21_2.FCStd.zip](https://github.com/user-attachments/files/17598062/BSplineChamfer0_21_2.FCStd.zip)

The problem seems a litle bit related to the base bspline form, because in some cases the inner edge chamfer is working and the outer not - in other base-bsline geometries, the outer is working and the inner not.

I'm aware, the length of the chamfer is need to be shorter than the "wall" thinkness - but i can not see a rule, in witch cases the chamfer is not posible (and ends in the error BRep_API: command not done).

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * Manipulator 1.4.9
  * 3D_Printing_Tools
  * fasteners 0.3.50
  * BOLTSFC
  * Assembly4 0.12.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-11-01:
Confirmed using (two different OCC versions):

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.48
  * Assembly4 0.50.15
  * A2plus 0.4.68
```

and

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.48
  * Assembly4 0.50.15
  * A2plus 0.4.68

```

Syres916, 2024-11-01:
Using some old AppImages, it would appear the regression was introduced before Jan 2024, if I get time this weekend I'll try to narrow it down further.

Syres916, 2024-11-03:
@it-guru I'm afraid you'll have to create a macro to create the two B-Splines in the Sketcb (forget the Pad and Chamfer) as I've got no idea how I'm supposed to replicate it, all the tools that I use to rebuild models by Python don't work with your 0.21 Sketch.

it-guru, 2024-11-04:
@Syres916 : I don't know what you mean. Why i should create tow B-Splines? - I have already a working sketch, on which i create the pad/extrution. How i can create a chamfer with a Macro?
I'm a litle bit confuced.

Syres916, 2024-11-04:
Ok, I'll try and explain. We both know the root cause of the chamfer issue is the code used to create the Sketch. I can even see the differences between them but unless I can actually _create_ the Sketch from scratch using Python it's going to be a very difficult job (for my skillset) to track down anywhere between August 4th 2023 and the end of Jan 2024 when the regression was introduced. Just to recompile one version of FreeCAD from August 2023 takes 1.5 hours on my PC.
So if you've never created a macro in FreeCAD, the first thing you need is to have the Python Console open in 0.21.x and just create the Sketch from scratch in the Gui and all the commands are written to the Python Console. When you've finished the sketch exit Edit mode and you can select all the lines back to the beginning of the Python Console, paste them into an empty text document and attach the text document here. This will then allow me to take that Python code and with some tweaks make a macro I can run after each compilation.

it-guru, 2024-11-04:
@Syres916 : Hm, "We both know the root cause ... is the code used to create the Sketch"  -> This is a news for me. 

The sketch is a sample for the problem. It's not the real outline i need. The real sketch have mutch more B-Spline-Nodes (>4000), so i haven't use it to describe the problem (because i think it's not handy for developers)

I have create the sketch by importing a svg from inkscape (imported to Draft-Workbench and then use the path to sketch function).

Does it makes sense, to create a marko from the (SVG-) import process?

codeotaku, 2026-09-05:
I think I have a fix for this (incidentally, when working on limit state chamfers) - this now complete without error.  Will be an OCC PR. 

[image]